FreeCAD/FreeCAD #17714: In Russian UI in the setup does not exist few settings.

dibor, 2024-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After choosing Russian language for UI few settings do not available in the preferences.
Can not to choose other languages, sets up display settings etc.
Preferences window do not save window size.
Possible to change UI language only by this way, Help -> Start
https://github.com/user-attachments/assets/adb6343e-6a12-4658-9045-590c7b3caccd

### Full version info

```shell
OS: macOS 15.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-06:
@kaktusus have you seen this before ?

Roy-043, 2024-11-06:
Similar: #12006.

kaktusus, 2024-11-07:
@dibor Have you tried doing a FreeCAD configuration reset?

maxwxyz, 2024-11-07:
Please try this in the latest weekly and try to restart in safe mode (Help>Restart in safe mode)

dibor, 2024-11-08:
> From "Max Wilfinger" ***@***.***>
> To "FreeCAD/FreeCAD" ***@***.***>
> Cc "dibor" ***@***.***>; "Mention" ***@***.***>
> Date 07/11/2024 9:22:31
> Subject Re: [FreeCAD/FreeCAD] In Russian UI in the setup does not exist
> few settings. (Issue #17714)

Hello.
Got latest weekly version - nothing changed.
Menu  "restart in safe mode" do not exists in the MacOS version

dibor, 2024-11-08:
> Please try this in the latest weekly and try to restart in safe mode (Help>Restart in safe mode)
Hello.
Got latest weekly version - nothing changed.
Menu  "restart in safe mode" do not exists in the MacOS version

dibor, 2024-11-08:
OS: macOS 15.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Help 1.0.3

maxwxyz, 2024-11-08:
That is not the latest weekly. Download the macOS version here: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

dibor, 2024-11-08:
Exact downloaded from this link!

dibor, 2024-11-08:
x64 version is run :)
But anyway, in russian UI preferences  like "Tools"->"Costomize" in english UI
Preferenses do not exists in the russion UI

maxwxyz, 2024-11-09:
Thanks for testing. Closing this as duplicate of https://github.com/FreeCAD/FreeCAD/issues/12006

maxwxyz, 2024-11-09:
@dibor here is a workaround: https://github.com/FreeCAD/FreeCAD/issues/12006#issuecomment-2179981942

closed 2024-11-09 by maxwxyz