FreeCAD/FreeCAD #17718: Part : Join bodies

ReneSci, 2024-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When Joining two (or more) Bodies (objects) the resulting Join is not replacing the position of the bodies in the tree :
[image]

Instead - of doing this:
[image]

The same for "Compound".

However - "Fusion" does it :
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: Danish/Denmark (da_DK) ]
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.65
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * MeshRemodel 1.9.23
  * OpenDark 2024.3.13
  * OpticsWorkbench 1.0.18
  * QuickMeasure 2022.10.28
  * sheetmetal 0.5.3
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-07:
Is it different than in the stable 0.21.2 version?

mwganson, 2024-11-08:
It looks like the connect object is not inside the App::Part container in the first image.  If you expand it you will probably see the objects nested inside the connect and inside the App::Part container.  This should be the expected behaviour I would think.  If you drag/drop the connect into the App::Part then you will see the expected behavior.

ReneSci, 2024-11-08:
> It looks like the connect object is not inside the App::Part container in the first image. If you expand it you will probably see the objects nested inside the connect and inside the App::Part [...]

Yes, that's right, but why does FreeCAD not place the resulting "Connect" inside App::Part where the two bodies came from by itself - that would be more logical. I.e. "Connect" replaces "Body1" and "Body2" under App::Part

ReneSci, 2024-11-08:
> Is it different than in the stable 0.21.2 version?

As far I remember - it's the same

mwganson, 2024-11-09:
Yes, that is a sensible statement.  If both of the child objects are in the same app::part then the new connect should go into that same app::part.  Alternatively, if there is an active app::part, then put the new connect into it, and otherwise if no active app::part, then put it as a top level object in the tree.  The latter probably being much simpler to implement.  This should hold true for all new objects created, except for part design where they must go into the active body.  This should be consistent behavior throughout FreeCAD I would say, then the users can become accustomed to the behavior.

ReneSci, 2024-11-27:
Why has this one been closed ?

luzpaz, 2024-11-27:
@ReneSci it's still open. (You might be confused by the linked issue which is unrelated)

ReneSci, 2024-11-27:
ok

ReneSci, 2024-12-04:
Just rested the latest nightly build:
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: Danish/Denmark (da_DK) ]
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.65
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * MeshRemodel 1.9.23
  * OpenDark 2024.3.13
  * OpticsWorkbench 1.0.18
  * QuickMeasure 2022.10.28
  * sheetmetal 0.5.10
```
The following works as I would expect:
Fusion:
[image]

Cut:
[image]

Common:
[image]

Section:
[image]

However - these does not:
-----------------------------------------------------

Connect:
[image]

Boolean-Fractments:
[image]

Clice:
[image]

Compound:
[image]

ReneSci, 2025-05-27:
Just tested the latest nightly build:
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41866 (Git)
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US) [ OS: Danish/Denmark (da_DK) ]
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/
Logical DPI/Physical DPI/Pixel Ratio: 96/141.951/1
Installed mods: 
  * A2plus 0.4.65
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * MeshRemodel 1.9.23
  * OpenDark 2024.3.13
  * OpticsWorkbench 1.0.18
  * QuickMeasure 2022.10.28
  * sheetmetal 0.7.22
  * workfeature
```

But the above "errors" still exists...

[Connect_TEST.zip](https://github.com/user-attachments/files/20452718/Connect_TEST.zip)

ReneSci, 2025-08-26:
Please let it stay open. I will test it, however I am quite busy for the moment.
I guess it has not been changed since there are no comments here.

Maybe I am the only one bothered with this.....

ReneSci, 2026-03-24:
Still relevant:

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260318 (Git shallow)
Build date: 2026/03/18 23:42:42
Build type: Release
Branch: grafted,grafted
Hash: 7886ebae105b99f176cddbfc62a106ab06a6d021
Python 3.11.15, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Danish/Denmark (da_DK)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
Logical DPI/Physical DPI/Pixel Ratio: 96/141.951/1
Installed mods: 
  * A2plus 0.4.68
  * CfdOF 1.34.5
  * Defeaturing Workbench 1.2.9
  * Optics Design Workbench 0.7.3
  * Manipulator Workbench 1.6.3
  * Mesh Remodel 1.11.0
  * OpenDark 2024.3.13
  * Optics 1.2.2
  * Quick Measure 2022.10.28
  * SheetMetal Workbench 0.8.4
  * toSketch 1.0.1
  * workfeature

```