FreeCAD/FreeCAD #17719: FEM color legend labels - numbers disappear

jurpoc, 2024-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

My deepest congrats for the amazing work done to develop FreeCAD. I am running some non-linear FEM simulations and I have some problems with data representation.

FEM color legend labels disappear after saving a document.

Situation before saving a document.
[image]

and after saving a document.
[image]

If the document is closed and opened again the legend labels reappear.

This happens for all simulations I run (Analysis type: Static).

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39127 (Git) AppImage
Build type: Release
Branch: main
Hash: 1b1e355b9b8c22ae8ca01c383f6b57cdac8a7c04
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Render 2024.9.25
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-06:
cc @FEA-eng

jurpoc, 2024-11-07:
Yes, Finite Element Method (FEM) is more related to the maths behind. I should wrote Finite Element Analysis (FEA).

luzpaz, 2024-11-07:
@FEA-eng should this be closed as a dupe ?

FEA-eng, 2024-11-08:
> @FEA-eng should this be closed as a dupe ?

Yes, I think so.

luzpaz, 2024-11-08:
Closed as a duplicate of #11665

closed 2024-11-08 by luzpaz