FreeCAD/FreeCAD #17721: [Sketcher] - Crash (SIGSEGV) - Sketcher.Constraint call

huguesdpdn-aerospace, 2024-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

# Problem
When using Sketcher.Constraint with invalid arguments, FreeCAD crashes with the following stacktrace:

```
FreeCAD_weekly-builds-39127-conda-Linux-x86_64-py311.AppImage 3221686410755.py 
FreeCAD 1.1.0, Libs: 1.1.0devR39127 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Unknown exception while processing file: 3221686410755.py 
XXX lineno: 68, opcode: 151
XXX lineno: 73, opcode: 151
XXX lineno: 194, opcode: 151
XXX lineno: 45, opcode: 149
SystemError: unknown opcode
XXX lineno: 45, opcode: 149
SystemError: unknown opcode
XXX lineno: 45, opcode: 149
SystemError: unknown opcode
XXX lineno: 45, opcode: 149
SystemError: unknown opcode
XXX lineno: 45, opcode: 149
SystemError: unknown opcode
XXX lineno: 45, opcode: 149
SystemError: unknown opcode
Program received signal SIGSEGV, Segmentation fault.
# 0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7fb5d1645320]
# 1  /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x1df8) [0x7fb5d39e72f8]
# 2  /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7fb5d3a0bc1f]
# 3  0x7fb5d6379c37 in PySide::SignalManager::callPythonMetaMethod(QMetaMethod const&, void**, _object*, bool) from /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/./libpyside2.cpython-311-x86_64-linux-gnu.so.5.15+0x97
# 4  /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/./libpyside2.cpython-311-x86_64-linux-gnu.so.5.15(+0x1fe15) [0x7fb5d6384e15]
# 5  /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libQt5Core.so.5(+0x1dfff1) [0x7fb5d1bdfff1]
# 6  0x7fb5d1be257a in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libQt5Core.so.5+0x28
# 7  0x7fb5d1bd9474 in QObject::event(QEvent*) from /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libQt5Core.so.5+0x5a
# 8  0x7fb5d2769fca in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libQt5Widgets.so.5+0xd8
# 9  0x7fb5d43694e9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libFreeCADGui.so+0xc9
#10  0x7fb5d1bc3158 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libQt5Core.so.5+0x8e
#11  0x7fb5d1bf2dec in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libQt5Core.so.5+0x250
#12  /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libQt5Core.so.5(+0x1f348a) [0x7fb5d1bf348a]
#13  /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/./libglib-2.0.so.0(+0x5b1d5) [0x7fb5ce1101d5]
#14  /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/./libglib-2.0.so.0(+0x5e377) [0x7fb5ce113377]
#15  /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7fb5ce113a8e]
#16  0x7fb5d1bf35af in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libQt5Core.so.5+0x59
#17  0x7fb5d1bc0dda in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libQt5Core.so.5+0x188
#18  0x7fb5d1bc4ea5 in QCoreApplication::exec() from /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libQt5Core.so.5+0xfb
#19  0x7fb5d42c8454 in Gui::Application::runApplication() from /tmp/.mount_FreeCAKpEMdN/usr/bin/../lib/libFreeCADGui.so+0x8a4
#20  /tmp/.mount_FreeCAKpEMdN/usr/bin/freecad(+0x78b7) [0x6334a3f508b7]
#21  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7fb5d162a1ca]
#22  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7fb5d162a28b]
#23  /tmp/.mount_FreeCAKpEMdN/usr/bin/freecad(+0x7c2a) [0x6334a3f50c2a]
```

# Expected
Expecting an Expcetion to be thrown , an error message but not a crash

# How to reproduce:
Please find in my next comment a file in attachment to reproduce the issue. Just execute:
`./freecad 3221686410755.py`

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (LXQt/Lubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39127 (Git) AppImage
Build type: Release
Branch: main
Hash: 1b1e355b9b8c22ae8ca01c383f6b57cdac8a7c04
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

huguesdpdn-aerospace, 2024-11-06:
I cannot attach Python file, so here is the file in raw:
```
import re
import sys
import time

class Piece_3221686410755():
    def __init__(self):
        self.__piece_vendor   = "castorama"
        self.__piece_isbn     = 3221686410755
        self.__piece_prefix   = ""
        self.__document_name  = str(self.__piece_vendor.upper()) + "_" + str(self.__piece_isbn)
        self.__document_class = None
        self.__in_freecad     = True
        self.__importImports()
        self.__computePieceInstanceInCurrentDocument()
        self.__create()

    def isThisScriptRunInFreeCAD(self):
        return self.__in_freecad

    def getDocumentClass(self):
        return self.__document_class

    def getDocumentName(self):
        return self.__document_name

    def getVendor(self):
        return self.__piece_vendor.upper()

    def getISBN(self):
        return self.__piece_isbn

    def getPieceNamePrefix(self):
        return self.__piece_prefix

    def __importImports(self):
        try:
            import FreeCAD
            self.__in_freecad = True
        except ModuleNotFoundError:
            self.__in_freecad = False
            print("[WARNING]: This script seems to not be run in FreeCAD")
            pass

    def __computePieceInstanceInCurrentDocument(self):
        pieces_with_same_isbn_list = []
        if self.getDocumentClass() is None or self.getDocumentClass().Objects is None or len(self.getDocumentClass().Objects) == 0:
            self.__piece_instance = 0
        else:
            for freecad_object in self.getDocumentClass().Objects:
                if hasattr(freecad_object, "TypeId") and freecad_object.TypeId == "App::Part" and freecad_object.Label.startswith(self.getVendor() + "_" + str(self.getISBN()) + "_"):
                    id_to_extract = None
                    try:
                        id_to_extract = int(freecad_object.Label[len(self.getVendor() + "_" + str(self.getISBN()) + "_"):])
                    except:
                        continue
                    else:
                        if id_to_extract is not None and isinstance(id_to_extract, int):
                            pieces_with_same_isbn_list.append(id_to_extract)
            pieces_with_same_isbn_list.sort()
            id_expected = 0
            while id_expected in pieces_with_same_isbn_list:
                id_expected = id_expected + 1;
            self.__piece_instance = id_expected
        self.__piece_prefix = self.getVendor() + "_" + str(self.getISBN()) + "_" + str(self.__getPieceInstanceInCurrentDocument())

    def __getPieceInstanceInCurrentDocument(self):
        return self.__piece_instance

    def __create(self):
        self.__document_class = FreeCAD.ActiveDocument
        if self.getDocumentClass() is None:
            self.__document_class = App.newDocument(self.getDocumentName())
            Gui.setActiveDocument(self.getDocumentName())
            App.setActiveDocument(self.getDocumentName())
        else:
            self.__document_name = getDocumentClass().getName()
        main_part = self.getDocumentClass().addObject("App::Part", self.getPieceNamePrefix())
        plate_body = self.getDocumentClass().addObject("PartDesign::Body", self.getPieceNamePrefix() + "_Plate_Body")
        main_part.addObject(plate_body)
        plate_body_sketch = plate_body.newObject("Sketcher::SketchObject", self.getPieceNamePrefix() + "_Plate_Sketch")

        plate_body_sketch_point_bottom__left = Part.Point(App.Vector( 0,   0, 0))
        plate_body_sketch_point_top_____left = Part.Point(App.Vector( 0, 150, 0))
        plate_body_sketch_point_top____right = Part.Point(App.Vector(75, 150, 0))
        plate_body_sketch_point_bottom_right = Part.Point(App.Vector(75,   0, 0))

        plate_body_sketch_geom_bottom__left  = plate_body_sketch.addGeometry(plate_body_sketch_point_bottom__left, True)
        plate_body_sketch_geom_top_____left  = plate_body_sketch.addGeometry(plate_body_sketch_point_top_____left, True)
        plate_body_sketch_geom_top____right  = plate_body_sketch.addGeometry(plate_body_sketch_point_top____right, True)
        plate_body_sketch_geom_bottom_right  = plate_body_sketch.addGeometry(plate_body_sketch_point_bottom_right, True)

        plate_body_sketch_line___left = Part.LineSegment(App.Vector( 0,   0, 0), App.Vector( 0, 150, 0))
        plate_body_sketch_line____top = Part.LineSegment(App.Vector( 0, 150, 0), App.Vector(75, 150, 0))
        plate_body_sketch_line__right = Part.LineSegment(App.Vector(75, 150, 0), App.Vector(75,   0, 0))
        plate_body_sketch_line_bottom = Part.LineSegment(App.Vector(75,   0, 0), App.Vector( 0,   0, 0))

        plate_body_sketch_geom___left = plate_body_sketch.addGeometry(plate_body_sketch_line___left)
        plate_body_sketch_geom____top = plate_body_sketch.addGeometry(plate_body_sketch_line____top)
        plate_body_sketch_geom__right = plate_body_sketch.addGeometry(plate_body_sketch_line__right)
        plate_body_sketch_geom_bottom = plate_body_sketch.addGeometry(plate_body_sketch_line_bottom)

        #vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
        #This line is THE line that makes FreeCAD crash
        plate_body_sketch.addConstraint(Sketcher.Constraint("Coincident", plate_body_sketch_line___left, 2, plate_body_sketch_line____top, 1))
        #^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

piece = Piece_3221686410755()
App.activeDocument().recompute()
```

huguesdpdn-aerospace, 2024-11-06:
Just to be clear, I perfectly know that this line is **INCORRECT**
`plate_body_sketch.addConstraint(Sketcher.Constraint("Coincident", plate_body_sketch_line___left, 2, plate_body_sketch_line____top, 1))`

the **CORRECT** line should be:
`plate_body_sketch.addConstraint(Sketcher.Constraint("Coincident", plate_body_sketch_geom___left, 2, plate_body_sketch_geom____top, 1))`

Notice that the local variables name are incorrect, I replaced in the names `line` by `geom`

I just passed the `Sketch` class instead of the `Geometry` class (almost the same variable name, I copied/pasted the wrong variables name, my bad).
The issue is that, even if the error is obvious, FreeCAD should had a safety measure to avoid the crash and even warn the user about this error.

Syres916, 2024-11-07:
This is a regression, using 0.21.x there was an error raised (it certainly could have been written to be more user helpful) but there's no seg fault. The error was:

```
11:02:38  Traceback (most recent call last):
  File "/home/john/Data/FreeCAD/Macro/00TestConstraintCrash.py", line 109, in <module>
    piece = Piece_3221686410755()
            ^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/Data/FreeCAD/Macro/00TestConstraintCrash.py", line 15, in __init__
    self.__create()
  File "/home/john/Data/FreeCAD/Macro/00TestConstraintCrash.py", line 106, in __create
    plate_body_sketch.addConstraint(Sketcher.Constraint("Coincident", plate_body_sketch_line___left, 2, plate_body_sketch_line____top, 1))
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'TypeError'>: Invalid parameters: ('Coincident', <Line segment (0,0,0) (0,150,0) >, 2, <Line segment (0,150,0) (75,150,0) >, 1)
Constraint constructor accepts:
-- empty parameter list
-- Constraint type and index
```

If I can at least narrow down when the regression was introduced, I'll report back later this afternoon/evening.

Syres916, 2024-11-07:
Well, glad I compiled the latest main build because I get the same error and no seg fault using:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Curves 0.6.51
  * Assembly4 0.50.15
  * A2plus 0.4.68
```

So is this a Conda build packaging issue or am I just lucky in the libraries I'm using???

huguesdpdn-aerospace, 2024-11-07:
I have an even weirder behavior .....

Using the official image (`official 21.2` from https://www.freecad.org/downloads.php ) under **Linux** : **Infinite loop**

```
$> ./FreeCAD-0.21.2-Linux-x86_64.AppImage 3221686410755.py 
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: Ne peut ouvrir le fichier d'objet partagé: Aucun fichier ou dossier de ce nom (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: iris
MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: Ne peut ouvrir le fichier d'objet partagé: Aucun fichier ou dossier de ce nom (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: iris
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: Ne peut ouvrir le fichier d'objet partagé: Aucun fichier ou dossier de ce nom (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
Unknown exception while processing file: 3221686410755.py 
No valid GL context found!
(qt.qpa.backingstore) composeAndFlush: QOpenGLContext creation failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
No valid GL context found!
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
...
```

Using the latest weekly image (`conda 39127` from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds ) under **Linux** : **SEGFAULT (CRASH)**
```
$> ./FreeCAD_weekly-builds-39127-conda-Linux-x86_64-py311.AppImage 3221686410755.py 
FreeCAD 1.1.0, Libs: 1.1.0devR39127 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Unknown exception while processing file: 3221686410755.py 
XXX lineno: 237, opcode: 151
SystemError: unknown opcode
XXX lineno: 68, opcode: 116
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x710663645320]
#1  /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libpython3.11.so.1.0(+0x22b0cc) [0x710665a2b0cc]
#2  /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libpython3.11.so.1.0(+0x22a9f0) [0x710665a2a9f0]
#3  0x7106663b0fe1 in Gui::PythonCommand::isActive() from /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libFreeCADGui.so+0x1d1
#4  0x7106663b5dff in Gui::Command::testActive() from /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libFreeCADGui.so+0x1ef
#5  0x7106663b6352 in Gui::CommandManager::testActive() from /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libFreeCADGui.so+0x22
#6  0x710666850299 in Gui::MainWindow::_updateActions() from /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libFreeCADGui.so+0x6a9
#7  /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libQt5Core.so.5(+0x1dffbf) [0x710663bdffbf]
#8  0x710663be257a in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libQt5Core.so.5+0x28
#9  0x710663bd9474 in QObject::event(QEvent*) from /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libQt5Core.so.5+0x5a
#10  0x710664769fca in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libQt5Widgets.so.5+0xd8
#11  0x7106663694e9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libFreeCADGui.so+0xc9
#12  0x710663bc3158 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libQt5Core.so.5+0x8e
#13  0x710663bf2dec in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libQt5Core.so.5+0x250
[... 6 lines cut ...]
#20  0x710663bc4ea5 in QCoreApplication::exec() from /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libQt5Core.so.5+0xfb
#21  0x7106662c8454 in Gui::Application::runApplication() from /tmp/.mount_FreeCAgkolBA/usr/bin/../lib/libFreeCADGui.so+0x8a4
#22  /tmp/.mount_FreeCAgkolBA/usr/bin/freecad(+0x78b7) [0x6023697eb8b7]
#23  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x71066362a1ca]
#24  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x71066362a28b]
#25  /tmp/.mount_FreeCAgkolBA/usr/bin/freecad(+0x7c2a) [0x6023697ebc2a]
```

Using the official image (`official 21.2` from https://www.freecad.org/downloads.php ) under **Windows (Portable AND Installer)** : **Python error in console**
```
Add here logs under Windows
```

[... cut ...] .0dev` from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds ) under **Windows** : **SEGFAULT (CRASH)**
```
Add here logs under Windows
```

I don't have access to other computers but I am interested if some can test.
Until then, let's assume that:
- it crashes only under Ubuntu 24.04 for the weekly build
- it goes into an infinite loop under Ubuntu 20.04 for the official release
- it just logs a Python error on any version on Windows

Anyone has a UNIX OS (any) and test it ?

luzpaz, 2025-08-22:
Since it's a crash bug, I've assigned it as a Blocker.

PaddleStroke, 2025-09-26:
So what happens here is that when you do : 

`Sketcher.Constraint("Coincident", line1, 2, line2, 1)`

You are feeding Constraint ctor the pointer to the Part.LineSegment. So what happens is that `ConstraintPy::PyInit` does not check the type of the input and so it parse the pointer as an int, which results in a garbage int. Which somewhere down the line crashes.

closed 2025-09-26 by chennes