FreeCAD/FreeCAD #17722: Sketcher: Geometry snaps to nearest element instead using the input value.

Panda631, 2024-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Open the sketcher.
Create a circle with D=29mm;
Create a second circle with D=30mm (input to on view field)

Result: 2 circles with 29mm.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.27
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.8.4
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-03-10:
The problem is that when you type 30, it updates the drawing to be at this position. Which generates a tangent autoconstraint (which is replaced with a equality). And because of the auto-constraint, the diameter becomes redundant. So it is not added.

I think autoconstraints should not be generated if the user use the OVP. Appears simple but's it's quite tricky actually. I'll check this out

PaddleStroke, 2025-03-10:
@abdullahtahiriyo if you happen to have not disabled your notifications and want to check this out.
This comes from 
```
    bool finish()
    {
        if (this->isState(SelectMode::End)) {
            unsetCursor();
            resetPositionText();

            try {
                executeCommands();

                if (sugConstraints.size() > 0) {
                    generateAutoConstraints();

                    beforeCreateAutoConstraints();

                    createAutoConstraints();
                }
            }
```
Where you generated autoconstraints before applying the OVP's constraints. Which generate this current issue.

I am failing to see why you did it this way. Why not first adding the OVP's constraint since they should be prioritary? Then adding autoconstraints only if they are not conflicting/refundant? If a user wants a 30mm diameter circle he wants that and probably don't care about autoconstraint.

closed 2025-03-17 by chennes