FreeCAD/FreeCAD #17727: {Feature] Define a MIME type for FCStd files and submit it to IANA

yorikvanhavre, 2024-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FCStd files have no official MIME type. They either inherit from .zip files and are seen as zip files, or, when the OS allows it, defined by FreeCAD as `application/x-extension-fcstd ` (ex. in https://github.com/FreeCAD/FreeCAD/blob/main/src/XDGData/org.freecad.FreeCAD.xml ).

We should:

- Decide either if we want to keep that "generic" MIME or suggest a more meaningful one (ex. That includes "FreeCAD" in the name)
- Register that new name to the IANA. Instructions at https://www.iana.org/assignments/media-types/media-types.xhtml - There is quite a lot of instruction to read.. :) That might be a task for the FPA.

Issue suggested by @pierreporte 

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-07:
@pierreporte FYI regarding new file types.

pierreporte, 2024-11-07:
After the discussion on Discord (with @kadet1090), we found that a good MIME type could be `application/vnd.freecad+suffix` or `application/x.freecad+suffix`. The `vnd` prefix requires registration while the `x` prefix doesn’t. Because the new file format is likely going to be FCStd with a formalized structure, only one MIME type is needed. However to make it easier for file system to show the difference, a suffix can be used. Here is a list based on the current state of the discussion:

* `application/vnd.freecad` for current FCStd (to be renamed playground or sandbox)
* `application/vnd.freecad+part` for parts designed with Part or PartDesign
* `application/vnd.freecad+part-sheetmetal` for parts designed with the Sheet Metal workbench (should it be merged into FreeCAD)
* `application/vnd.freecad+assembly` for assemblies
* `application/vnd.freecad+drawing` for drawings (a set on one or more pages)
* `application/vnd.freecad+bim-freestyle` for BIM projects using unstructured workflow
* `application/vnd.freecad+bim-strict` for BIM projects following the IFC structure

mwganson, 2024-11-07:
Is this case-sensitive?  There should be consideration to move to .fcstd or at least leave that open as a future option.  Dragging and dropping a .fcstd file (or opening from file menu) does work, at least on windows and on a self-compiled ubuntu version.  Presumably it works on the AppImage and on the mac versions.  It might require a separate registration for the .FCStd and for the .fcstd, but I would suggest both be done if possible, or at least consideration given.

yorikvanhavre, 2024-11-08:
Blender uses `application/x-blender`. Wouldn't that be more appropriate? (application/x-freecad)

pierreporte, 2024-11-09:
@yorikvanhavre [According to Wikipedia](https://en.wikipedia.org/wiki/Media_type#Unregistered_tree), the dashed version has been deprecated for nearly 30 years (!) and can’t be registered since 2013. FreeCAD can’t justify historic usage to make an exception. Blender must have been using `application/x-blender` since the beginning, while FreeCAD haven’t even started.

Are we keeping `text/plain` for *.FCMat?

@mwganson No idea but, in any case, I plan to recommend lowercase extensions alongside the new file format.

maxwxyz, 2024-11-17:
@davesrocketshop ^ regarding *.FCMat

kadet1090, 2024-11-17:
Ideally it should be something like `application/x.feeecad.material+yaml` to signify that it is specific yaml file.

pierreporte, 2024-11-17:
Then the other should be `application/vnd.freecad.std+suffix` (as a reference to FCStd).

luzpaz, 2025-03-18:
Should this move to the FPA repo ?

oerkel47, 2026-08-24:
On Cinnamon, it is very annoying when you can only open .FcStd as a zip archive or open zip archives with FreeCad.

The MIME Type would fix that if I understand correctly.

See this old but still relevant post: https://forums.linuxmint.com/viewtopic.php?t=398045