FreeCAD/FreeCAD #17728: [Part Design] Transparent parts turn opaque after every new feature

jes, 2024-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I often make my complex parts 50% transparent so that I can see the features on the inside.

This has worked fine previously.

I'm using FreeCAD 1.0rc2 now, and every time I add a new Part Design feature, it resets the transparency of my object to turn it opaque again.

Example: https://www.youtube.com/watch?v=c3tfnlODH_w

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38641 +165 (Git) Snap 1162
Build type: Release
Branch: tag: 1.0rc2
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jes, 2024-11-07:
It looks like it is resetting the material completely. If I change the colour on the Body, or select a different material from the list, it gets reset to default opaque grey every time.

maxwxyz, 2024-11-07:
Already fixed in latest weekly

closed 2024-11-07 by maxwxyz