FreeCAD/FreeCAD #17730: PartDesign: PolarPattern artifact

Pendelf, 2024-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I made Subtractive Helix, then when I try to make Polar Pattern I get this

[image]

https://github.com/user-attachments/assets/754742d2-33cf-4975-9c95-02eff685c995

[PolarPattern Problem.zip](https://github.com/user-attachments/files/17662564/PolarPattern.Problem.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.29
  * PieMenu 1.9.1
  * sheetmetal 0.5.3
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-11-07:
Fixed by switching off the _SubtractiveHelix_ `Refine` property.

FlachyJoe, 2024-11-07:
OCC geometric core is very sensible with helix.
I rant about automatic refinement if you work with helix you should disable it in PartDesign preferences.

Pendelf, 2025-08-27:
I believe it is still relevant

[image]

hisepacker, 2026-08-25:
I’m seeing what appears to be a similar PolarPattern artifact while designing a lamp frame in FreeCAD.

The model is a precisely dimensioned structural frame with repeated geometry arranged around a central axis. The individual source geometry appears correct, but when I create a polar replication/pattern, an unexpected triangular/wedge-shaped face appears on one instance of the repeated geometry.

The artifact is not present in the original feature and only appears after the polar pattern is created.

In the same file, I have also experienced some related Boolean robustness problems when working with precisely aligned cutter solids, including failed/null Boolean results near coincident or closely aligned faces. I experimented with Boolean fuzzy tolerance values around 0.001 and with Refine enabled, but the behavior has not been consistently resolved.

The PolarPattern artifact is the most visually obvious problem: a large triangular region appears where no corresponding geometry exists in the source feature.

I am using this model to create a real, dimensionally accurate lamp frame for fabrication/3D printing, so the geometry is intentionally constrained rather than approximately positioned.

I can provide the `.FCStd` file, screenshots, and exact reproduction steps if they would be useful. I can also reduce the model to a smaller reproducer if needed.

This may be another example of the issue described here, but with substantially different source geometry.