FreeCAD/FreeCAD #17731: FreeCAD - Размер списка последних файлов

h0tab1ch, 2024-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

в настройках портабле версии не меняется Размер списка последних файлов там любую цифру ставишь жмешь применить перезагружаешь-обновляешь программу но все равно показывает только пустой и 4 последних

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-11-07:
Please add English translation to the description and change the title to English.

maxwxyz, 2024-11-07:
Please create the issue in English or translate. Please test on the latest weekly whttps://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-11-07:
> Version: 0.21.2.33771 (Git)

Please use RC2 (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2). It should be fixed. 

If not, here are some Start Workbench related bugs that are already open related to Recent Files https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aopen+is%3Aissue+label%3A%22Mod%3A+Start%22

Closing ticket.

And please open tickets in english...thank you.

closed 2024-11-07 by luzpaz