FreeCAD/FreeCAD #17732: [1.0][Gui] Greedy selection mode not working correctly

shaise, 2024-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It looks like there is a bug in the greedy selection mode:

Create a cube.
Enter greedy selection mode by console command:
`Gui.Selection.setSelectionStyle(Gui.Selection.SelectionStyle.GreedySelection)`
Select several edges.
Now click one of the selected edges (in order to deselect it) - problem: all other selections disappear.
However they are still considered as selected. If you issue the command:
`Gui.Selection.getSelectionEx()[0].SubElementNames`
A list of the still selected edges appear.
 

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * fasteners 0.5.29
  * kicadStepUpMod 11.3.9
  * Manipulator 1.5.7 (Disabled)
  * OpenTheme 2024.9.1
  * PieMenu 1.8.1 (Disabled)
  * Render 2024.9.25
  * SelectorToolbar (Disabled)
  * sheetmetal 0.5.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

shaise, 2024-11-07:
Here is a video description:

https://github.com/user-attachments/assets/fbc779a2-a717-407d-8b06-667aa7dee207

maxwxyz, 2024-11-07:
@shaise could this be related to (the already fixed issue): https://github.com/FreeCAD/FreeCAD/issues/16920 
Please try on the latest weekly

shaise, 2024-11-08:
@maxwxyz , Yes! this is definitely solved in the latest build, and also backported to 1.0.
Closing this issue.

closed 2024-11-08 by shaise