FreeCAD/FreeCAD #17733: FreeCAD что такое .FCBak

h0tab1ch, 2024-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD создает иногда файлы копии проекта с расширением  .FCBak которые потом сам не видит, я понимаю что это типа бэкап проекта но мусор получается и он накапливается.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-07:
Please create the issue in English or translate. Please test on the latest weekly whttps://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-11-07:
Please use RC2 https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2
Issue is solved there. Closing

closed 2024-11-07 by luzpaz