FreeCAD/FreeCAD #17734: FreeCad 1.0.0RC2  Assembly Stability (joints)

jimsybeldon, 2024-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Specifically, I have tested multiple ways to assemble parts in an assembly.  (A fireplace frame, mantel, and shiplap).     Whether one assembly or an assembly of subassemblies,  eventually the solver bogs down to be unusable.    This occurs when I keep the part very primitive mean just rectangular parts and use 3 planes to attach to a fixed first part.

I also have added holes to join a part to it's mating part using fixed joints to greatly simplify computation.   That did help.  However when I tested the assembly by adding a new hole feature to a part,  the containing assembly scrambled positions.  (this feature was added only, not used for a joint.  It was a new feature not a modification of an existing feature or that features sketch)

This should have no effect on the assembly but it did break the system of joints.   At this point I can predictably get either slow solver issues, or assembly instabilities.   I have been able to get to this problem in different parts/assemblies that were all built from the ground up.  (Meaning no use of common parts for other assembly attempts.)

The same approaches to part and assembly concepts where tested in Solidworks, Inventor, and Onshape.  All can be done successfully with good speed and stability.

The addition of a new feature to a part causing assembly errors would be a TNP issue in the past, but that I solved, correct?
[image]

### Full version info

```shell
FreeCad1.0.0RC2
![image](https://github.com/user-attachments/assets/2e0ac90f-3c0a-40c4-ace2-6e53a20b4596)
```

### Subproject(s) affected?

Assembly

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-07:
Please test on the latest weekly as there have been some fixes in assembly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
If there is still an issue, please share an example file and steps how to reproduce

drwho495, 2024-11-07:
im still having TNP issues making this wb unusable atp (on the latest dev build)

jimsybeldon, 2024-11-07:
OK all,

Will get the latest build.

TNP had made assembly workbenches -  in the sense any cad system creates
assemblies -  impossible in freecad.   (I realize you know that.  I am
operating on the premise that TNP was behind us.)

 I appreciate what the assembly workbench has evolved into.

I will be working on blowing up assemblies in the meantime.  I just have
difficulties proving to myself that I am not running over the same old TNP
issues.

Any advice on how I can provide useful assembly workbench feedback in light
of TNP?  Let me know.  The commingled nature of TNP and Assy WB problems
mean I got problems without useful feedback for the devs......

Thanks,

jimsybeldon, 2024-11-08:
Toponaming issues that are open that will effect Assy WB

https://github.com/FreeCAD/FreeCAD/issues/17412
https://github.com/FreeCAD/FreeCAD/issues/17266
https://github.com/FreeCAD/FreeCAD/issues/16256
https://github.com/FreeCAD/FreeCAD/issues/15486
https://github.com/FreeCAD/FreeCAD/issues/14720
https://github.com/FreeCAD/FreeCAD/issues/14643

jimsybeldon, 2024-11-08:
New info related to sluggish response for your consideration:   I have created the situation in various attempts.  I will create a model with this issue but for this moment the report view explains the situation quite well:

12:40:57  0 <Assembly> AssemblyObject.cpp(176): Solve failed: iterNo > iterMax
12:41:39  25.6887 <Assembly> AssemblyObject.cpp(176): Solve failed: iterNo > iterMax
12:41:39  42.1418 <Assembly> AssemblyObject.cpp(176): Solve failed: iterNo > iterMax
12:46:15  317.736 <Assembly> AssemblyObject.cpp(176): Solve failed: iterNo > iterMax
12:47:05  344.83 <Assembly> AssemblyObject.cpp(176): Solve failed: iterNo > iterMax
12:47:05  368.237 <Assembly> AssemblyObject.cpp(176): Solve failed: iterNo > iterMax

I was aware that once I get a number of distance joints greater than 100,  it gets much worse quickly.    100 joints is not unusual for a machine assembly.   zip model will follow soon.

jimsybeldon, 2024-11-08:
Latest build is being used:

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.61
  * Assembly3 0.12.0
  * Assembly4 0.50.6
  * Help 1.0.3

Regarding max interations:    
[Model for MaxInterations Issue.zip](https://github.com/user-attachments/files/17683606/Model.for.MaxInterations.Issue.zip)

You will be able to create maximum iterations error by attempting to place move parts into that assembly model:
-  has been tested doing a recompute for every new joint added.   That helps but the  max iterations issues gets worse with more joints and eventually even rebuild after each joint will fail you.
- I recommend that you add another part to the assembly, add distance joints for that another part until it is fully defined.  Do not recompute between adding the required joints.  (3 joints to fully lock down all D.O.F)   You will have difficulty even achieving that.

Regarding the TNP issue:
[Model for TNP testing.zip](https://github.com/user-attachments/files/17683697/Model.for.TNP.testing.zip)

This the same model as above, but is provide purely to establish that a TNP issue will effect this model.     Here I simply added a hole feature to the "chimney side shiplap" part.    While it did not immediately blow up the model, a naming issue is seen is the report view:

14:03:00  <Exception> TopoShapeExpansion.cpp(2019): Invalid shape name ?Face3
14:03:00  Assembly: Invalid shape name ?Face3
14:03:00  <Exception> TopoShapeExpansion.cpp(2019): Invalid shape name ?Face3
14:03:00  Assembly: Invalid shape name ?Face3
14:03:05  <Exception> TopoShapeExpansion.cpp(2019): Invalid shape name ?Face3
14:03:05  Assembly: Invalid shape name ?Face3
14:04:06  <Exception> TopoShapeExpansion.cpp(2019): Invalid shape name ?Face3
14:04:06  Assembly: Invalid shape name ?Face3

[image]

Other notes:  as stated originally,  I had attempted to address the max iterations issue by providing mating hole features and use "fixed" joints in lieu of "3 faces to lock all DOF per part".   While that should help the assembly tremendously,  the TNP issue precedes this.  It is not possible as things stand now.

jimsybeldon, 2024-11-08:
On the tnp testing model I intend that on of you should add features or alter an existing part, the see if assembly errors from TNP are created.  The model as provided works since the parts are all original and no way altered.  It is yours to break.

jimsybeldon, 2024-11-16:
#17734
Puttiing this in the dead letter office.  Not on anybody's radar.   More fundamental issues

closed 2024-11-16 by jimsybeldon