FreeCAD/FreeCAD #17735: Surface "Fill boundary curves" gives error "Curves are disjoint" in FreeCAD 1.0.0RC2 and FreeCAD 1.0.0dev37100

Xargas, 2024-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[Checks](https://forum.freecad.org/viewtopic.php?t=5236):
1) Latest development version used -> yes
2) Is Bug -> yes
3) FreeCAD forum -> check
4) Reproducible -> yes
5) Version info -> see below
6) Include all error messages -> Text, Screenshot
7) Terminal, FreeCAD Log -> Screenshot, no info there
8) Cool and patient -> 🫡

The bug occured while redo-ing the solution from the forum:
https://forum.freecad.org/viewtopic.php?p=790611#p790787

Problem:
1) Open FreeCAD 1.0.0RC2 or 1.0.0dev37100
2) Create a new file
3) Create a new Sketch on XY plane using Sketcher containing:
-- two lines and one circle
-- the two lines form a 90° edge joint
-- the circle is concave towards the 90° edge
-- all three form a face form a compact manifold
4) Switch to the Surface workbench and create a "Fill boundary curves" object.
-- Add the three edges in order: arc, line, line
-- The other two possible orders seem to work. As this is a triangle it should not be an issue regarding the order, like in other cases.
5) Try to finish surface creation, get the error message "Curves are disjoint"

[image]

### Full version info

```shell
Buggy version dev:

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Austria (en_AT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CurvedShapes 1.0.12
  * Curves 0.6.47
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Context: https://forum.freecad.org/viewtopic.php?p=790611

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-08:
Thanks for this report. We need more improvement to Surface WB for sure.

wwmayer, 2024-11-09:
I don't think it's a regression but I guess it has never worked because in v0.21 it fails, too. However, instead of three edges you can select two edge to create the surface.

I assume the problem is in `GeomFillSurface::createBSplineSurface` where the line segments and arc of circle are converted into B-Splines in order to be able to use OCC's `GeomFill_BSplineCurves`. If the B-Spline from the arc of circle is not a real NURBS curve then there is a loss of accuracy and this could mean that there is a tiny gap between the three B-Spline curves.

Xargas, 2025-01-05:
I justed tried it again with FreeCAD 1.0.0 Release, but the issue is still there.
Simplest example possible:
[ExampleForError.zip](https://github.com/user-attachments/files/18310014/ExampleForError.zip)
The two working order of selections are shown, the third one causing the issue is not create-able.

When I tried to do a copy&paste and changed the order of edges in the "Boundary List", the error also occurs:
[image]

```
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Austria (en_AT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CurvedShapes 1.0.12
  * Curves 0.6.47
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * ThreadProfile 1.92.0
```

tarman3, 2025-01-05:
Confirm
<details>
  <summary>FreeCAD 1.1.0dev.39748 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39748 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: e977eb49e37ef836f695809263fa1f56a17cc657
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.34; sheetmetal 0.6.13; Silk 0.1.5; Curves 0.6.53
```

</details>

Shkolik, 2025-02-09:
Following... Surface WB needs help for sure.

Xargas, 2025-02-20:
I made myself familiar with the code of the Surface WB and identified that this path is the failing path:

1. FreeCAD: App::DocumentObjectExecReturn* GeomFillSurface::execute()
2. FreeCAD: createBSplineSurface(aWire);
3. FreeCAD: aSurfBuilder.Init(curves[0], curves[1], curves[2], fstyle);
4. OCCT: Init(C1, C2, C3, C4, Type);
5. OCCT: Arrange(C1, C2, C3, C4, CC1, CC2, CC3, CC4, Tol);

The arrange function does not guarantee a correct result in some cases when the fourth line degenerates to a point. This is the case we have here, has OCCT introduces this "virtual line" in (4) to close the loop.

The case when it works is, when the "C4 virtual line" is added exactly in the order as shown in the picture.
But depending on the order of edges (this issue here), the "C4 virtual line" may be added "between C1 and C2", so exactly the other way round.

```
//                      CC3
//                  ----->-----
//                 |           |
//                 |           |
//                 |           |
//             CC4 ^           ^ CC2
//                 |           |
//                 |           |
//                  ----->-----
//                   CC1 = C1
```

I am currently only able to build FreeCAD with the existing LibPack.
When I get it to build with a local built OCCT I can try to create a fix.

I document my findings here in case someone has already such a working build setup and can test a fix.

Xargas, 2025-02-20:
I was able to use a local built OCCT library with FreeCAD.

A fixed arrange function would be one that first searches for degenerated lines (=points).

```
static Standard_Boolean Arrange(const Handle(Geom_BSplineCurve)& C1,
                                const Handle(Geom_BSplineCurve)& C2,
                                const Handle(Geom_BSplineCurve)& C3,
                                const Handle(Geom_BSplineCurve)& C4,
                                Handle(Geom_BSplineCurve)&       CC1,
                                Handle(Geom_BSplineCurve)&       CC2,
                                Handle(Geom_BSplineCurve)&       CC3,
                                Handle(Geom_BSplineCurve)&       CC4,
                                const Standard_Real              Tol)
{
  Handle(Geom_BSplineCurve) GC[4];
  Handle(Geom_BSplineCurve) Dummy;
  GC[0] = Handle(Geom_BSplineCurve)::DownCast(C1->Copy());
  GC[1] = Handle(Geom_BSplineCurve)::DownCast(C2->Copy());
  GC[2] = Handle(Geom_BSplineCurve)::DownCast(C3->Copy());
  GC[3] = Handle(Geom_BSplineCurve)::DownCast(C4->Copy());

  Standard_Integer i, j;
  Standard_Boolean Trouve;

  for (i = 1; i <= 3; i++)
  {
    Trouve = Standard_False;

    // search for a degenerated line = point, which would match first
    for (j = i; j <= 3 && !Trouve; j++)
    {
      if (GC[j]->StartPoint().Distance(GC[j]->EndPoint()) < Tol)
      {
        // this is a degenerated line, does it match the last endpoint?
        if (GC[j]->StartPoint().Distance(GC[i - 1]->EndPoint()) < Tol)
        {
          Dummy  = GC[i];
          GC[i]  = GC[j];
          GC[j]  = Dummy;
          Trouve = Standard_True;
        }
      }
    }

    // if no degenerated line matched, try an ordinary one
    if (!Trouve)
    {
      for (j = i; j <= 3 && !Trouve; j++)
      {
        if (GC[j]->StartPoint().Distance(GC[i - 1]->EndPoint()) < Tol)
        {
          Dummy  = GC[i];
          GC[i]  = GC[j];
          GC[j]  = Dummy;
          Trouve = Standard_True;
        }
        else if (GC[j]->EndPoint().Distance(GC[i - 1]->EndPoint()) < Tol)
        {
          GC[j]  = Handle(Geom_BSplineCurve)::DownCast(GC[j]->Reversed());
          Dummy  = GC[i];
          GC[i]  = GC[j];
          GC[j]  = Dummy;
          Trouve = Standard_True;
        }
      }
    }

    // if still non matched -> error
    if (!Trouve) {
      std::cout << "Error at" << i << std::endl;
      return Standard_False;
    }
  }

  CC1 = GC[0];
  CC2 = GC[1];
  CC3 = Handle(Geom_BSplineCurve)::DownCast(GC[2]->Reversed());
  CC4 = Handle(Geom_BSplineCurve)::DownCast(GC[3]->Reversed());

  return Standard_True;
}
```

This would at least fix this issue. But we have to ask OCCT if there might be side effects.
I think this is worth a related issue at OCCT3D. I currently do not have an account there, so I at least document my finding/potential bugfix here.

Xargas, 2025-02-20:
Open-Cascade-SAS/OCCT#386

Xargas, 2025-04-18:
Info:
The patch was accepted in OCCT: https://github.com/Open-Cascade-SAS/OCCT/issues/386
When FreeCAD is updating to OCCT >=8.0.0 this issue should be fixed: https://github.com/Open-Cascade-SAS/OCCT/discussions/499

Xargas, 2025-10-26:
The bug is still present. I tested it in

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Austria (en_AT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Gridfinity 0.9.4
  * Manipulator 1.5.7
  * Render 2024.12.15
  * ThreadProfile 1.92.0
```

The fix "GeomFill updated with fix of incorrect arrangement of Degenerated BSpline curve https://github.com/Open-Cascade-SAS/OCCT/pull/396" is in OCCT 8.0.0 RC1 https://github.com/Open-Cascade-SAS/OCCT/discussions/499. So at least this version needs to be included in the FreeCAD release to close this bug.

FEA-eng, 2026-05-05:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260429 (Git shallow)
Build date: 2026/04/29 07:07:16
Build type: Release
Branch: grafted,grafted
Hash: a785d40ac4c37f559b0b82fbaf9bf7dc68a468e4
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

Xargas, 2026-11-13:
still an issue
tested today on 

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 1.1.4.20260928 (Git shallow)
Build date: 2026/09/28 13:46:34
Build type: Release
Branch: (HEAD detached at 4fd3bf320)
Hash: 4fd3bf320d9566a27e60069fc8387448aaa3a094
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.4, OCC 7.8.1
Locale: German/Austria (de_AT)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.933/1
Installed mods: 
  * Curved Shapes 1.0.15
  * Curves workbench 0.6.72
  * Fasteners Workbench 0.5.62
  * FreeCAD AI 0.33.0-alpha
```

the order of selection defines if the result is working or not - but the result shall not depend on the order