FreeCAD/FreeCAD #17737: CAM:  Helix drilling random path

sslupsky, 2024-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When generating a tool path using the Helix Drilling operation the tool path can be random.  I have attached an illustration that shows the result of the Helix operation.

1.  You can see that in some cases, there is no tool path.  I have determined that in those situations it appears the cylinder has more than one face.  This appears to be caused by the polar pattern causing the generation of an additional face when replicating the holes.
2. Another peculiar behaviour is the tool path appears to have a different radius in some cases.  In these situations the cylinder appears to have only one face but for some reason the tool path is different.  This example was generated with 12mm holes and a 6mm Endmill.
3. Another problem appears to be the Starting Radius property.  In the second illustration, if I use a 3.175mm endmill and set the Starting Radius to 6mm - 3.175mm/2 = 4.4125mm a second tool path appears at the edge of the hole that should not be there.

[image]

[image]

### Full version info

```shell
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git)
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * ThreadProfile 1.85.0
  * kicadStepUpMod 10.19.4
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sslupsky, 2024-11-07:
@maxwxyz The closest issue I could find when I looked earlier is #12601.

This model was created to try to recreate the issue in a simple drawing.  This example has four shapes.  The shape furthest from the center is a circle where the edge was split using the split tool in the sketcher.  These features are not picked up by the Helix Drill operation as compared to the simple circle drawn closest to the middle.

[cam hole profiling debug.FCStd.zip](https://github.com/user-attachments/files/17668280/cam.hole.profiling.debug.FCStd.zip)

The other issues I will have to upload a more complex drawing.

sslupsky, 2024-11-07:
Here is a more complex model illustrating the first two issues.
[helix toolpath problem.FCStd.zip](https://github.com/user-attachments/files/17668331/helix.toolpath.problem.FCStd.zip)

sslupsky, 2024-11-07:
I updated the first example by adding another helix operation.  The second operation illustrates my third point above.  I also manually added an edge of one of the outer circles to the first helix operation.  This appears to illustrate the second point above (helix with different radius).

[cam hole profiling debug 2.FCStd.zip](https://github.com/user-attachments/files/17668415/cam.hole.profiling.debug.2.FCStd.zip)

luzpaz, 2024-11-07:
Any CAD folk available to comment ?

luzpaz, 2025-05-31:
@sslupsky any chance you can re-check this and see what is still problematic and what isn't on the latest 1.1dev ?