FreeCAD/FreeCAD #17738: Cannot type circumflex in expression editor

wdubbert, 2024-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I cannot type the circumflex character in the expression editor for exponentiation (FreeCAD ignores the input). It works when I copy it from somewhere else and paste it in the editor. I was able to reproduce the issue on another machine (also running Linux Mint).

[image]

My keyboard layout is german (QUERTZ).

### Full version info

```shell
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * A2plus 0.4.68
```

### Subproject(s) affected?

Expressions

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mwganson, 2024-11-08:
In the python console enter:

`ord("^")`

What do you get for both types (typed in and pasted in)?  Should be 94.

wdubbert, 2024-11-08:
For the pasted in one I get 94, but I cannot type it there either.

mwganson, 2024-11-09:
There are couple more things to try:

1. Ctrl+Shift+U followed by 5e
2. Caret, then follow with space

1 is to allow unicode input, 5e being the unicode for the caret.  2. is in case the caret key is a dead key used to accent other characters, and pressing space should output the caret, hopefully.

wdubbert, 2024-11-10:
Unfortunately both ways didn't work. When I type the Caret followed by a space, I only get the space. Ctrl+Shift+U does not seem to register, when I type 5e afterwards I get the characters '5' and 'e', when I hold Ctrl+Shift+U and type 5e (which gives me a caret in gedit) I get nothing. Tested it in both the Expression Editor and the Python Console and got the same behaviour.

xtemp09, 2024-11-10:
@wdubbert, are you trying to enter a caret or a circumflex? Do other _Qt_ application work fine?

wdubbert, 2024-11-10:
I am not quite sure what the difference between the caret and the circumflex is. What I am trying to do is an exponentiation and I am using the top left key on my QWERTZ-Keyboard (ASCII-Code 0x5e). I just switched my keyboard layout to QWERTY and tried Shift+6 which seems to work fine.

I also just checked Filezilla (which according to the Qt wiki also utilizes Qt) and have no problems there.

luzpaz, 2025-01-25:
So what's the diagnosis here?

MiqCG, 2025-02-13:
Same problem here. I have just discovered this issue thanks to @xtemp09 , talking about this other [issue](https://github.com/FreeCAD/FreeCAD/issues/12173)

I am using a QWERT**Y** keyboard with Catalan and Spanish layout (ç and ñ letters) in FreeCAD, in English, but Catalan as system locale. There's only one ^ symbol in my keyboard that can be used to exponentiate or to accentuate. I think that what I have in common with @wdubbert is at least the Debian base. The ^ keystroke is working fine in, for example, SimpleScreenRecorder and CopyQ software. My system is up-to-date, although I admit I have a problem with updating initramfs-tools for a very long time (probably more than a year) due to a excessively small boot partition.

Testing in the console:
- pasting `ord("^")` results in 94 but I cannot type the ^symbol in there either. Same as @wdubbert 
- Caret, then follow with space renders just the space
- Ctrl+Shift+U followed by 5e results in
  File "<input>", line 1
    5e
    ^
SyntaxError: invalid decimal literal

Problem occurs in Spreadsheets and Expression editor.

Surprisingly enough, it happens with appimages 0.21.1, 1.0.0, and various weekly-builds (pre and post 1.0.0). Even with freshly deleted .config, .cache. and .local/share FreeCAD folders.
Definitely I didn't have this problem in the 26th of August 2024 when I reported to the other issue #12173. I cannot be sure what was the situation in between.

`OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 103/100.707
`

MiqCG, 2025-02-15:
I have installed FreeCAD 1.0.0 via [flatpak](https://flathub.org/apps/org.freecad.FreeCAD) and the problem is not present there. So, can it be a specific problem of appimage and its dependencies?

luzpaz, 2025-12-10:
> https://github.com/FreeCAD/FreeCAD/issues/17738#issuecomment-2656655120

@johnocfii would you mind testing on macOS to see if this is not Linux-exclusive ?