FreeCAD/FreeCAD #17739: UI/UX: Add an Ondsel-esque preference pack option Vanilla install dialog

luzpaz, 2024-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Many of the Ondsel users will be migrating back to FC and will complain about the UI. Maybe we can add an option for them to choose an Ondsel-esque preference pack (that triggers the addon manager to download a yet to be created pref pack) that will load the ondsel UI/UX pref pack that was used (I believe it was a combination of the external openthemes stylesheet and openpreferences addon.

### Full version info

```shell
1.0
```

### Subproject(s) affected?

Core

### Anything else?

I'm aware that we are about to release 1.0 and this probably won't make it in. But it could possibly into 1.0.1

The proposal involves:  
* packaging openthemes and openpreferences in to a pref pack
* updating the First Run dialog to add an Ondsel-esque option
* writing logic that auto-downloads this preference pack when the the Ondsel-esque option is chosen
* making sure the preference pack installs correctly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-11-08:
Adding a button on the start screen for everyone which references a product that only some used sounds like a strange route to me.

Wouldn't it be better if ondsel wrote a blog post how to configure freecad to look the same instead?

And doesn't the options transfer so users should get the same look out of the box anyway?

maxwxyz, 2024-11-08:
@obelisk79 so basically like your preference pack?

obelisk79, 2024-11-08:
> @obelisk79 so basically like your preference pack?

Sound like the proposal is literally my preference pack.

luzpaz, 2024-11-08:
> Adding a button on the start screen for everyone which references a product that only some used sounds like a strange route to me.

True. But here's the logic:  
1. Ondsel did have a following and those folks may be experiencing some disorientation coming back to the classic UI
2. This would only apply to 1.0 (we could remove it in the next major release or minor release)

Another idea:
Have a generic image that indicates an alternative stylesheet/theme and link to the addon manager Stylesheet/Theme section so users can explore other themes/preference packs

fidoriel, 2024-12-05:
I tried ondsel last year, but was not happy. But with freecad 1.0 I decided it is time to ditch Fusion360 for new projects. ondsel UI/UX is closer to F360. I followed a blog post on the internet to get FreeCad to look like ondsel, but was not able to match everything, it was not the same. I think it would benefit the initial setup and overall user retention.

luzpaz, 2024-12-05:
@prokoudine did a really great write-up[^1]: **How to make FreeCAD look and work like Ondsel ES**.
Within it is a step to download the source file preference pack[^2] from the ondsel-es-feedstock[^3] directly.

[^1]: https://librearts.org/2024/11/freecad-to-ondsel-es-look/
[^2]: https://librearts.org/2024/11/freecad-to-ondsel-es-look/#install-the-preference-pack
[^3]: https://github.com/Ondsel-Development/ondsel-es-feedstock/blob/88b5e6d7bc5a1020607c8a6739bd1e178e327973/recipe/branding/Ondsel-defaults.cfg#L4

luzpaz, 2026-09-22:
I guess this feature wasn't popular. I'll risk the theory that most Ondsel suers migrated to AstoCAD. So this FR is obsolete now.

Closing.

closed 2026-09-22 by luzpaz