FreeCAD/FreeCAD #17744: Sketcher: Angle constrain on x-axis can't selected anymore

Panda631, 2024-11-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Place an angle constrain on top of the x-axis. 
No way to select and change it anymore.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.27
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.8.4
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-08:
Same as https://github.com/FreeCAD/FreeCAD/issues/12756 
See https://github.com/FreeCAD/FreeCAD/issues/12756#issuecomment-2365165358
@FlachyJoe FYI

Syres916, 2024-11-08:
How can this regression which I've just confirmed to have been generated by a commit between:

Good:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36082 (Git) AppImage
Build type: Release
Branch: main
Hash: 7a025e4e60a607bdbdd1599627ce13bb8eeda071
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Curves 0.6.51
  * Assembly4 0.50.15
  * A2plus 0.4.68
```

Bad:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git) AppImage
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Curves 0.6.51
  * Assembly4 0.50.15
  * A2plus 0.4.68

```

be closed as well as #12756 , one of them must be kept _Open_

maxwxyz, 2024-11-08:
Then we can leave this open. @FlachyJoe you've mentioned in the other one that you might have a fix?

Syres916, 2024-11-08:
The first thing that needs making clear is which commit back in late Feb/early March caused the regression, there is absolutely no mention of this information in #12756 so I'm doing a Git Bisect now to clarify the commit, hopefully it's compilable.

FlachyJoe, 2024-11-25:
@Syres916 did the git bisect give results?

luzpaz, 2025-10-25:
Is this still relevant ?

Panda631, 2026-09-01:
Just a friendly reminder.
Issue is still valid.

tritao, 2026-09-01:
I think this should be fixed by https://github.com/FreeCAD/FreeCAD/pull/32261

closed 2026-09-01 by Lgt2x