FreeCAD/FreeCAD #17745: IFC: Unable to move Hybrid IFC project to different disk location

semhustej, 2024-11-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When project is saved as a Hybrid project, IFC file is created. When both files are moved to a different folder, and the FCstd file is opened, it shows error:

Steps to reproduce:

- Start a new file
- Switch to strict IFC mode (IFC project is created)
- Switch back to hybrid IFC mode
- Save the file (both FCstd and IFC file)
- Close the file (save upon closing)
- Move  (do not copy) both FCstd and IFC file to a different folder on the disk
- Open the FCstd file
- Result is error, it seems FC can not read the IFC file

Error message:
```
10:31:31  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 54, in onChanged
    self.edit_schema(obj, obj.Schema)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 182, in edit_schema
    ifcfile = ifc_tools.get_ifcfile(obj)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 409, in get_ifcfile
    ifcfile = ifcopenshell.open(project.IfcFilePath)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\__init__.py", line 161, in open
    return file(f)
           ^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\file.py", line 264, in __init__
    raise exc(msg)
<class 'OSError'>: Unable to open file for reading
10:31:31  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 61, in onChanged
    self.edit_attribute(obj, "Name", obj.Label)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 159, in edit_attribute
    ifcfile = ifc_tools.get_ifcfile(obj)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 409, in get_ifcfile
    ifcfile = ifcopenshell.open(project.IfcFilePath)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\__init__.py", line 161, in open
    return file(f)
           ^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\file.py", line 264, in __init__
    raise exc(msg)
<class 'OSError'>: Unable to open file for reading
10:31:31  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 59, in onChanged
    self.edit_attribute(obj, prop)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 159, in edit_attribute
    ifcfile = ifc_tools.get_ifcfile(obj)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 409, in get_ifcfile
    ifcfile = ifcopenshell.open(project.IfcFilePath)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\__init__.py", line 161, in open
    return file(f)
           ^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\file.py", line 264, in __init__
    raise exc(msg)
<class 'OSError'>: Unable to open file for reading
10:31:31  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 59, in onChanged
    self.edit_attribute(obj, prop)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 159, in edit_attribute
    ifcfile = ifc_tools.get_ifcfile(obj)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 409, in get_ifcfile
    ifcfile = ifcopenshell.open(project.IfcFilePath)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\__init__.py", line 161, in open
    return file(f)
           ^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\file.py", line 264, in __init__
    raise exc(msg)
<class 'OSError'>: Unable to open file for reading
10:31:31  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 59, in onChanged
    self.edit_attribute(obj, prop)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 159, in edit_attribute
    ifcfile = ifc_tools.get_ifcfile(obj)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 409, in get_ifcfile
    ifcfile = ifcopenshell.open(project.IfcFilePath)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\__init__.py", line 161, in open
    return file(f)
           ^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\file.py", line 264, in __init__
    raise exc(msg)
<class 'OSError'>: Unable to open file for reading
```

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

BIM

### Anything else?

With this behavior, it is impossible for a user to send a hybrid project to a different location as you can see in this forum thread: https://forum.freecad.org/viewtopic.php?t=91960

Possible solutions:

- have an option (and/or overview) to change the path to the IFC file
[image]

- use relative path to the IFC file (or an option to use relative path)

And:
- add more informative error message if FC is unable to find the IFC file

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-08:
@yorikvanhavre FYI

yorikvanhavre, 2024-11-08:
Indeed. At the moment the path to the IFC file is hard-coded. We should detect the file on file opening and offer a dialog to reconciliate if the file has been moved.