FreeCAD/FreeCAD #17748: Pocket and MillFace not cutting to full depth in latest dev build

dresco, 2024-11-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the latest dev build, both Pocket_Shape and MillFace operations are missing the final (full depth) pass.

If the same file is opened in in latest 1.0 build, and operations are recomputed, then the full depth is machined as expected.

Attaching a simplified file that demonstrates this issue.

### Full version info

```shell
Failing;

OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Unknown
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * btl 0.9.9
  * FeedsAndSpeeds 0.5.0
  * ThreadProfile 1.92.0
  * kicadStepUpMod 11.4.1

Working;

OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.0.0RC2.39069 (Git)
Build type: Unknown
Branch: releases/FreeCAD-1-0
Hash: 9b3329ff9c648b9ac7f954ba60f97ea0b7c0b22f
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * btl 0.9.9
  * FeedsAndSpeeds 0.5.0
  * ThreadProfile 1.92.0
  * kicadStepUpMod 11.4.1
```

### Subproject(s) affected?

CAM

### Anything else?

[cam profile test.zip](https://github.com/user-attachments/files/17679294/cam.profile.test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GeneGH, 2024-11-23:
The problem is more that partial step-downs are no longer processed. If the start depth, finish depth, and step down have a nice integral relationship the full depth is processed. If not, only full step-downs are processed and any remaining depth is skipped.

Connor9220, 2024-11-25:
I think this bug is also in v1.0.0 as well. I've not tested with RC4 to see if it's present in it.

dresco, 2024-11-25:
@Connor9220 Can confirm the same issue in RC4 & v1.0.0, at least with the attached test file.

@GeneGH I believe the missing pass in the MillFace operation is a full depth cut? (Start depth 0mm, final depth - 10mm, step down 5mm).

maxwxyz, 2024-12-04:
@jbaehr could you take a look at this? Seems to happen after you PR.

dresco, 2024-12-04:
Just as another data point, a workaround for the MillFace operation was mentioned on Discord. Changing the boundary shape to Face Region instead of Boundbox does give the final pass. Not an option for pocket though..

The message logged when the final pass is skipped is `<Path.Area> Area.cpp(1993): Discard empty section`

jbaehr, 2024-12-04:
> @jbaehr could you take a look at this? Seems to happen after 8362c5af8b8d9be0f9216efe11bedcc9f2f295a9

Wow, that's a huge changeset! I think it'll be more efficient if the original authors of this take a look.

This commit seems to originally come from #17119, which landed on main on October 21th  as cf8ad6637351819dfa07d49cfd65d511fe14a084 and was backported to v1.0 with #17325 on November 11th as 8362c5af8b8d9be0f9216efe11bedcc9f2f295a9. Indeed, when this issue here was created, on November 8th, it was reported that only main was affected, explicitly not v1.0. And on November 25th @Connor9220 confirmed it for v1.0 as well.

@dresco in the original #17119 you'll find the change as 51 individual commits. Maybe bisecting them can bring us closer to a solution?

dresco, 2024-12-04:
> @dresco in the original #17119 you'll find the change as 51 individual commits. Maybe bisecting them can bring us closer to a solution?

Makes sense, I'll try bisecting the @AIRCAP fixBooleans branch tomorrow. Sadly, my dev laptop is old and crusty, and takes about an hour per iteration to rebuild ;( Maybe Eric will have some ideas in the meantime..

dresco, 2024-12-05:
New bisect on the fixBooleans branch has identified [af065b2](https://github.com/FreeCAD/FreeCAD/pull/17119/commits/af065b20a466e8f11797becba757f76bf220d75e) as the first bad commit, "introduce wrapper for BRepAlgoAPI_BooleanOperation and derivatives, set sensible default Fuzzyness for boolean operations"

jbaehr, 2024-12-05:
When configuring new fuzzyness for boolean operations to 0.0 (I guess this means: "back to the old behaviour"), as described here https://github.com/FreeCAD/FreeCAD/issues/17543#issuecomment-2442456260, recomputing both operations bring back the final depth pass in the test file. (compiled from Yesterday's main-branch at a97677ed2f0ee4875321f3043782cf77e3756cc6 on macOS/arm64 and OCCT 7.7.2)

On the other hand, using an official mac build of 0.21.1 with OCCT 7.6.3, recomputing the test file brings the final depth back only for the Pocket. For the FaceMill nothing changes, i.e. no final depth pass, but the following warnings in the console:
```
21:11:37  <Path.Area> Area.cpp(1429): hit bottom -0,0,1e-06
21:11:37  <Path.Area> Area.cpp(1429): hit bottom 0,0,1e-07
21:11:37  <Path.Area> Area.cpp(1559): Discard empty section
```

jbaehr, 2024-12-05:
I played a bit more with parameter BaseApp/Preferences/Mod/Part/Boolean/BooleanFuzzy. For bringing the final depth pass for the MillFace back, I have to go down to 0.001 (0.01 is too large). For the Pocket, a value of 8 is enough (9 is too large; [10 is the default](https://github.com/FreeCAD/FreeCAD/pull/17119/files#diff-309a23165363ed7a2463177cddc0d448529974530b0cc707e1e85b6d014bef5eR569)).

Also note that the currently applied tolerance for boolean operations is "this fuzziness times OCCT's 'confusion' circle (1e-6 IIRC) times the part's dimension in mm". So in contrast to previous FreeCAD versions, it's now basically a measure *relatively* to the part's size. And this size is derived from the part's bounding box, which [does include the poles of NURBS](https://opencascade.blogspot.com/2009/04/all-inclusive-bounding-box.html) (if based on geometry, not on a triangulated representation). And this means that these tolerances can grow unexpectedly large for curved shapes. That's not the case here, but something to keep in mind. So -- to be fair -- @AIRCAP's work fixed a ton of outstanding bugs. But it also opens some new can of worms it seems.

Connor9220, 2024-12-25:
What is the status of this?  This is a pretty major issue.  Makes Millface and Pocket very difficult to use.  I've resorted to using RC2 for now.

jbaehr, 2024-12-28:
> What is the status of this?

I don't work on this. My hope lies in @AIRCAP, who seems to have the deepest knowledge about this boolean tolerance stuff. I guess the additional tolerance ("fuzziness") introduced with #17119 became a property of the resulting solid and is now also used when libarea probes for how far it can step down and thus stops too early. This assumption is based on the log message "<Path.Area> Area.cpp(1993): Discard empty section" ([source](https://github.com/FreeCAD/FreeCAD/blob/7083362acb8529017b58ea03bf7416da0192b25b/src/Mod/CAM/App/Area.cpp#L1993)).

> I've resorted to using RC2 for now.

Have you tried setting the parameter "BaseApp/Preferences/Mod/Part/Boolean/BooleanFuzzy" to zero as described in the [comment above](https://github.com/FreeCAD/FreeCAD/issues/17748#issuecomment-2521339847)?

Connor9220, 2024-12-29:
> > I've resorted to using RC2 for now.
> 
> Have you tried setting the parameter "BaseApp/Preferences/Mod/Part/Boolean/BooleanFuzzy" to zero as described in the [comment above](https://github.com/FreeCAD/FreeCAD/issues/17748#issuecomment-2521339847)?

It was not clear to me that I could change this in the Parameters page.  I'm assuming I need to use a float value?  I tried this with .001 and facemill was still broken.  Set it to 0 and it looks to work.  Does that disable this new code?  What other things will this affect in FreeCAD?

closed 2025-02-14 by sliptonic