FreeCAD/FreeCAD #17749: No way to copy / paste object properties (placement)

maxwxyz, 2024-11-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is currently no way to copy properties from one object to a different object, e.g. placement values.

For example: Copying one object's placement properties and selecting a different object and pasting them so the placement property is transformed to the new object.

In the case of placement there is a workaround: Open the placement editor while the old part is selected, select the new object and hit apply.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * CfdOF 1.27.13
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.25
  * Rocket 4.0.0
  * sheetmetal 0.5.3
  * Silk 0.1.5
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mwganson, 2024-11-09:
DynamicData has a copy/set function that can copy a Placement or any other property to another object.  The arrows point in the direction of the data flow.

Copy = create a new property and set it to this value
Set = set an existing property to this value
Bind = set the value in an expression.
Break binding = clear the expression

I never thought to offer using the clipboard for this.  That could work to copy across different documents.

maxwxyz, 2024-12-29:
This would also be helpful for BIM: Copy a stairs object, paste all or selected attributes to a different stairs object (or any other object like wall, window,...) @yorikvanhavre FYI

kaiwas, 2025-02-25:
@maxwxyz  I would like to clarify that my proposal in the issue you closed is of a different nature. Your points imply copying and pasting properties between two objects.
My proposal is to create a tool for transferring multiple properties to multiple objects. For example, it could be at least a hundred windows, walls or holes. And most importantly, the objects can be completely different, and not just "Part". That is, for example, the height specified in the sketch can be transferred to a column

therefore, I believe that this is not a complete duplicate. Conceptually, this is a completely different approach.

pieterhijma, 2025-09-01:
I think this issue is more broad than the Part module. To distinct it from @22966, I would suggest to rename this issue to:

Gui: No way to copy / paste property values between objects

What is the expected outcome if the placement happens to be the result of an expression? Is only the value copied to the new object or is the expression copied as well? In the latter case, is the expression updated relative to the new object if possible or absolutely, meaning that it refers to the old object?

mwganson, 2025-09-02:
In DynamicData when copying / setting properties and copying by expression where there are ambiguities the user is presented with the text of the expression in an editable field.  Whether to copy by value or by expression can be set using a checkbox in the dialog.