FreeCAD/FreeCAD #17752: Enhancement put full path into title.

iplayfast, 2024-11-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've noticed that when I'm working on several iterations of a design, and have them saved in different places, it can become confusing as to which I'm actually working on. It would be helpful if the full path name were put into the titlebar. Or failing that, "..." plus the last 30 characters, eg ".../Documents/cad/gearbox/rev29/gearbox"

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Unknown
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * freecad.gears 1.0.0
  * 3D_Printing_Tools
  * CycloidGearBox
  * FeedsAndSpeeds 0.5.0
  * Ondsel-Lens 2024.1.22.01
  * ExplodedAssembly
  * Nodes 0.1.36
  * 3DfindIT 1.2.0
  * A2plus 0.4.63
  * MeshRemodel 1.9.22
  * lattice2 1.0.0
  * fasteners 0.5.10
  * Assembly3 0.12.2
  * Assembly4 0.50.7
  * POV-Ray-Rendering
  * PieMenu 1.2.7
  * CurvedShapes 1.0.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mwganson, 2024-11-09:
If you click on the document label and look in the data tab of the property view there is a file name property that will give that information.

If you want it in the title, here is how you can do this for yourself in a short macro:

```
mw = FreeCADGui.getMainWindow()
doc = FreeCAD.ActiveDocument
filename = doc.FileName
mw.setWindowTitle(filename)
```

It's only temporary until a new file is opened or probably if you switch documents.  You could probably add a document observer that would change the title with the active document changed, but I'm not 100% sure on that.

luzpaz, 2024-11-11:
@iplayfast can you make a screenshot mockup of what you're referring to ?

Syres916, 2026-01-09:
> Yes, this makes sense [@Syres916](https://github.com/Syres916) !

I did winder why this hadn't been done before and after testing multiple files open with spreadsheets and Techdraw pages I'm starting to understand why. Basically the tooltip is active only when that specific tab is clicked and it's active wherever the cursor is in the view. I'll look at the Status bar but that's already very overcrowded with the new hints.

iplayfast, 2026-02-27:
I tried out the macro suggestion mentioned by @mwganson 
```
mw = FreeCADGui.getMainWindow()
doc = FreeCAD.ActiveDocument
filename = doc.FileName
mw.setWindowTitle(filename)
```
Which works for about 1 millisecond. Then the title is replaced by the project name and freecad version which is really not that useful. The project name is already in the tab, and the tree, the file location is not so obvious, and the title is a nice long space to put it in.

The fact that it is replace right away byt the project name, shows that something already has this code implemented, I just want it to show more useful info. (IMHO).

leoheck, 2026-02-27:
Full path at the window title may work for you, which has a really huge resolution. I can see 1/8 of that picture you sent above. This should not be the default thing otherwise, it will create a mess on the screen for the normal users.

iplayfast, 2026-02-27:
If you see the original comment, I suggested that for long name they are shorted to some ... string. This is the same as almost every other program that I know of, Freecad is the exception. Word processors, spreadsheets and so on, all put the filename in the titlebar.

leoheck, 2026-02-27:
But what is the goal? If you put ellipsis at the beginning it will prioritize showing the filename that you already know. If you put that in the end it will hide the file name. 

Please, keep the window title simple as FreeCAD [Version]. Otherwise it will look a real mess. 

Having this information displayed on the status bar when hovering the document name would be enough. In addition having a tooltip with it when hovering the mouse over the tab could also work nicely without making the UI busy as hell.

leoheck, 2026-03-04:
It is so disgusting put the file name in the title bar. Status bar is longer since the font there is smaller. Also, the best would be just put it hidden in a tooltip when hovering the mouse over the Document Name and/or Tab Name.

Hope Designer team doesn't allow adding this thing on the title bar. This will make the whole window busy. If your OS has a busy UI then it is not others users problem. 

If this idea is added to the project it is better they put a non default settings for that thing.