FreeCAD/FreeCAD #17753: 1.0.rc2:  recomputation for migration  fails with 'Can not open font file...'

fbakan, 2024-11-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Opening a file containing 'draft-shapes' created with 0.21.2  in 1.0.rc2 works but  ... 'recomputation for migration purposes' .
fails with
'<class 'Part.OCCError'>: Can not open font file: /usr/share/fonts/truetype/...' 

of course the font files do exist on my computer.

example of the project:
[image]

### Full version info

```shell
OS: KDE Flatpak runtime (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 3d63fc6)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
```

### Subproject(s) affected?

None

### Anything else?

shortened errorlog from terminal-window:
```
~>flatpak run org.freecad.FreeCAD
FreeCAD 1.0.0, Libs: 1.0.0RC2R38806 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

_IceTransSocketUNIXConnect: Cannot connect to non-local host localhost.localdomain
_IceTransSocketUNIXConnect: Cannot connect to non-local host localhost.localdomain
Qt: Session management error: Could not open network socket
v1.0, Tico LP 1113, 'Verschmelzen'-, 'Ausrichten'-, 'Ausrichtungsreferenz'-, 'LinksBündigHalten'-, 'NeigungsWinkel'- und 'PassendSkalieren'- Eigenschaften hinzugefügt
v1.0, Tico LP 1113, geänderter 'Tracking' Eigenschaftstyp
v1.0, ECHANDO PA'LANTE, 'Verschmelzen'-, 'Ausrichten'-, 'Ausrichtungsreferenz'-, 'LinksBündigHalten'-, 'NeigungsWinkel'- und 'PassendSkalieren'- Eigenschaften hinzugefügt
v1.0, ECHANDO PA'LANTE, geänderter 'Tracking' Eigenschaftstyp
...
Traceback (most recent call last):
  File "/app/freecad/Mod/Draft/draftobjects/shapestring.py", line 140, in execute
    char = Part.makeWireString("L", obj.FontFile, 1, 0)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: Can not open font file: /usr/share/fonts/truetype/Utopia-Bold.ttf
Traceback (most recent call last):
  File "/app/freecad/Mod/Draft/draftobjects/shapestring.py", line 140, in execute
    char = Part.makeWireString("L", obj.FontFile, 1, 0)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: Can not open font file: /usr/share/fonts/truetype/NotoSans-Bold.ttf
...

```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-09:
Can you try this in the latest weekly version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

fbakan, 2024-11-09:
tried, but the appimages do not work here:

```
~/apps/freecad-weekly>./FreeCAD_weekly-builds-39141-conda-Linux-x86_64-py311.AppImage
FreeCAD 1.1.0, Libs: 1.1.0devR39141 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Authorization required, but no authorization protocol specified

This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, minimal, minimalegl, offscreen, vnc, webgl, xcb.

/tmp/.mount_FreeCAFPflcb/AppRun: Zeile 31:  5119 Abgebrochen             (Speicherabzug geschrieben) "${MAIN}" "$@"

```

I also had no luck when trying to compile myself. cmake fails with:
```

~/src/freecad-source/src/Build>cmake ../
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in CMakeLists.txt:
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at 3rdParty/lazy_loader/CMakeLists.txt:8 (SET_PYTHON_PREFIX_SUFFIX):
  Unknown CMake command "SET_PYTHON_PREFIX_SUFFIX".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.30)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!

```

luzpaz, 2024-11-11:
@fbakan can you redownload the latest weekly appimage and test (the appimage should work now)

fbakan, 2024-11-11:
tried, but the appimage still does not start, same problem

```
~/apps/freecad-weekly>./FreeCAD_weekly-builds-39153-conda-Linux-x86_64-py311.AppImage
FreeCAD 1.1.0, Libs: 1.1.0devR39153 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Authorization required, but no authorization protocol specified

This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, minimal, minimalegl, offscreen, vnc, webgl, xcb.

/tmp/.mount_FreeCAmLglma/AppRun: Zeile 31: 95098 Abgebrochen             (Speicherabzug geschrieben) "${MAIN}" "$@"

```

fbakan, 2024-11-13:
tried the new RC3 build from flatpak, but the problem still exists, appimages still don't work here

OS: KDE Flatpak runtime (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0RC3.39100 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 3c10471)
Hash: 3c1047103cc8960a92a90a03869a9370b80e6776
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion

maxwxyz, 2024-11-14:
There is a new weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds and RC4, could you please try again?

fbakan, 2024-11-14:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  Already tried, doesn't work here. Should I open an extra issue for 'Weekly builds don't work on tumbleweed?I'll try RC4 when it arrives on flathub.                                                                                                                                                                                                                                                                                                                        Von: ***@***.***: 14. November 2024 17:52An: ***@***.***: ***@***.***: ***@***.***; ***@***.***: Re: [FreeCAD/FreeCAD] 1.0.rc2:  recomputation for migration  fails with 'Can not open font file...' (Issue #17753)  
There is a new weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds and RC4, could you please try again?

fbakan, 2024-11-23:
Just tried again with the 1.0.0 installed from official openSUSE-Tumbleweed Repository
Now all works fine. No more errors when migrating.

closed 2024-11-23 by fbakan