FreeCAD/FreeCAD #17755: Part Design:  Additive Helix takes a very long time to generates a peculiar result

sslupsky, 2024-11-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have been struggling with the Additive Helix for a while.  It seems to have a couple problems:
1.  It takes a very long time to generate the helix
2. When it does, the result is often not expected

I have attached a demo here.  This illustrates the peculiar behaviour.  It appears as if the helix extends along the positive and negative vertical axis.

I find the angle property is not intuitive.  If the angle changes from a positive angle to a negative angle the direction of the helix along the axis is reversed similar to the Reversed property.  I am expecting the slope of the cone would change relative to the axis.

[image]

[helix problem.FCStd.zip](https://github.com/user-attachments/files/17682980/helix.problem.FCStd.zip)

### Full version info

```shell
OS: macOS 14.6.1
Architecture: x86_64
Version: 1.1.0dev.39127 (Git)
Build type: Release
Branch: main
Hash: 1b1e355b9b8c22ae8ca01c383f6b57cdac8a7c04
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sslupsky, 2024-11-09:
I had another thought about the amount CPU resources and I noticed that when applying the Helix Ramp Entry dressup it too takes a very long time to complete.  I wonder if these two issues are related in that they share some common code related to the generation of the helix?

maxwxyz, 2024-11-09:
Is it faster in the stable 0.21.2 version?
@FlachyJoe FYI

Syres916, 2024-11-09:
Most of the _time_ issue is because the OP has been changing the Angular Deviation and Tessellation settings to make models look good without taking the performance implications into account IMHO. OCC is only doing what it's being told and if it's asked to do high precision display then there has to be a CPU computation tradeoff. Max deviation of 0.25% and Max angular deviation of 10 degrees is what I use and find it to be a reasonable balance between aesthetics and performance.

Syres916, 2024-11-09:
@sslupsky is the attached what you were trying to achieve? Just remove the .zip from the filename.

[helix problem_Syres.FCStd.zip](https://github.com/user-attachments/files/17687418/helix.problem_Syres.FCStd.zip)

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.3.33862 (Git)
Build type: Release
Branch: 021_Python_3_12_And_Other_Fixes
Hash: f9bffbb8b811a4f6213ec829c164ae09f7af0e50
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Curves 0.6.51
  * Assembly4 0.50.15
  * A2plus 0.4.68
```

FlachyJoe, 2024-11-09:
> Most of the _time_ issue is because the OP has been changing the Angular Deviation and Tessellation settings to make models look good without taking the performance implications into account

:+1: 

I read a OCC guru about possible improvement in FreeCAD display mechanism (without more details). IMHO we should engage a work on this.

In my mind the optimal process will show a low polygon result very quickly to give to the user the possibility to continue their work and next refine the display at background. Therefore it implies a multithread process somewhere.

sslupsky, 2024-11-09:
> Most of the _time_ issue is because the OP has been changing the Angular Deviation and Tessellation settings to make models look good without taking the performance implications into account IMHO. [...]

I am aware of the tessellation settings and adjusted up and down to attempt to find an acceptable setting.  Unfortunately, what I found is the consequence is more significant than simply a visual display issue.  When you make a measurement the measurements are affected by the issue and thus measurements are dependent on these settings as well.

I have attached two illustrations showing this problem.  The first illustration is using the settings you suggested.  The second illustration is the settings I used.  The distance measurement between the two curves is different by about 4%.

0.25%, 10 degrees
[image]

0.2, 2.0 degrees
[image]

sslupsky, 2024-11-09:
> @sslupsky is the attached what you were trying to achieve? Just remove the .zip from the filename.

Similar but this model suffers from the same issue.  When you move the sketch shape closer to the vertical axis (close but not touching), the resulting helix goes wonky.  I think it may have something to do with using a Frenet curve.  It does not appear there is an option to turn Frenet on/off.  I think the plane of the sketch must be normal to the edge of the helix at the tip to avoid the distortion but there doesn't appear to be a straight forward way to do that.  I thought maybe starting from the bottom of the helix would help but there is no way to accomplish that because there seems to be a problem with the angle property.

[image]

sslupsky, 2024-11-09:
Follow up on my comment about having the sketch normal to the helix.  I have confirmed that is indeed what is going on.  If I change the plane of the sketch relative to the helix the shape is not distorted.  I haven't found a way to model it normal but I got close with trial and error.

[image]

FlachyJoe, 2025-09-14:
@FEA-eng I can't find any reason to reopen in the last comments. They're not directly related to the reported issue.

FlachyJoe, 2025-09-15:
> the performance issues

Due to a misconfiguration of tesselation so this issue is solved

> the distortion of the helix

Please open a well titled new issue

and 

> When you make a measurement the measurements are affected by the issue and thus measurements are dependent on these settings as well.

Please open a second issue mentioning Std_Measure. It's certainly not specific to PartDesign_Helix.

FlachyJoe, 2025-09-15:
> the time to create the helix is exceedingly long using the settings @Syres916 recommended.

I get the result in 3s. with the 0.5 deviation and 28.5° deflection default setting. Is it too much?