FreeCAD/FreeCAD #17757: measure distance  between two holes

Roberto8303, 2024-11-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

is it a problem or is it not possible to have the measurement between the centers of the two holes?

### Full version info

```shell
OS: Windows 10 build 19042
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/OpenDark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * Behave-Dark-Colors 0.1.1
  * CamScripts
  * ExplodedAssembly
  * fasteners 0.5.29
  * FeedsAndSpeeds 0.5.0
  * Freecad-Built-in-themes-beta 1.2.2
  * FreeCAD-themes 2024.7.24
  * OpenTheme 2024.9.1
  * PieMenu 1.8.0
  * ProDarkThemePreferencePack 1.0.0
  * QuickMeasure 2022.10.28
  * Render 2024.9.25
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-08:
> Version: 1.1.0dev.38923 (Git)

You're not using an up to date weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

maxwxyz, 2024-11-09:
Already fixed in the weekly

closed 2024-11-09 by maxwxyz