FreeCAD/FreeCAD #17761: Addon Manager errors

jwptrek, 2024-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Addon manager doesn't show known existing workbenches, specifically Rocket.  If I get it to appear through some settings, it doesn't appear properly.  Addon manager always gives the message Invalid version specified for tag version in Addon Trails: 2022-01.  It initially complained that git was disabled but it was enabled in the preferences.  When enabling, the report view changed font and gave multiple errors.  See error copy below.
  
20:34:01  "Empty widget item in QVBoxLayout 'verticalLayout'."
20:34:06  Git is disabled, skipping git macros
20:34:19  Invalid version specified for tag version in Addon Trails: 2022-01
20:34:34  Git is disabled, skipping git macros
20:34:40  Invalid version specified for tag version in Addon Trails: 2022-01
20:35:09  Invalid version specified for tag version in Addon Trails: 2022-01
20:35:38  DirectWrite: CreateFontFaceFromHDC() failed (Indicates an error in an input file such as a font file.) for QFontDef(Family="8514oem", pointsize=10, pixelsize=20, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("8514oem", lfWidth=0, lfHeight=-20) dpi=192

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

I previously had v0.20.1 installed.  I installed 0.21.2 after issues with updating a workbench from addon manager.  I tried uninstalling and reinstalling multiple times.  I also deleted the config files, without success.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jwptrek, 2024-11-09:
I had to select "show all" from the addon manager to see Rocket.  When trying to install Rocket, I got the below. 20:47:29  Traceback (most recent call last):
20:47:29    File "C:\Program Files\FreeCAD 0.21\Mod\AddonManager\AddonManager.py", line 822, in table_row_activated
20:47:29      self.packageDetails.show_repo(selected_repo)
20:47:29    File "C:\Program Files\FreeCAD 0.21\Mod\AddonManager\package_details.py", line 166, in show_repo
20:47:29      self.display_repo_status(self.repo.update_status)
20:47:29    File "C:\Program Files\FreeCAD 0.21\Mod\AddonManager\package_details.py", line 363, in display_repo_status
20:47:29      "<h1>"
20:47:29  TypeError: can only concatenate str (not "Version") to str
20:47:38  Traceback (most recent call last):
20:47:38    File "C:\Program Files\FreeCAD 0.21\Mod\AddonManager\AddonManager.py", line 949, in on_package_status_changed
20:47:38      self.packageDetails.show_repo(repo)
20:47:38    File "C:\Program Files\FreeCAD 0.21\Mod\AddonManager\package_details.py", line 166, in show_repo
20:47:38      self.display_repo_status(self.repo.update_status)
20:47:38    File "C:\Program Files\FreeCAD 0.21\Mod\AddonManager\package_details.py", line 363, in display_repo_status
20:47:38      "<h1>"
20:47:38  TypeError: can only concatenate str (not "Version") to str
/

jwptrek, 2024-11-09:
I uninstalled v0.21 and installed v0.20.2.  The package manager there told me the addon Rocket needed Freecad v1.0.  I tried v1.0rc2 and was able to get Rocket to install.  However, the Invalid version specified for tag version in Addon Trails: 2022-01 message still appears.  It was always in black text regardless of version.

maxwxyz, 2024-11-09:
please try the latest weekly version if there is still an issue: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

davesrocketshop, 2024-11-17:
Some new tags were added to the latest Rocket WB package.xml. Specifically:

`<freecadmin>1.0</freecadmin>`
`<depend type="addon" optional="true">CfdOF</depend>`

See issue #17921 for more explanation of the first tag