FreeCAD/FreeCAD #17762: Incorrect sweep result in case of B-spline path.

kaiwas, 2024-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The picture shows two objects. Circles on B-spline path.
The left one is obtained using Loft in Part Design (correct and you can even make a hollow object).
The right one is obtained using the Sweep tool in Part Workbench. (The geometry is broken and it is impossible to make a hollow object as a pipe)

[image]
[image]

The test file is attached. Remove the zip extension.

The result is the same in version 0.21.2

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39141 (Git) AppImage
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * addFC.backup1731047833.082574 0.7.4 (Disabled)
  * addFC 0.7.5
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-11-09:
I'll be honest, I've not searched for this in the over 2000 issues but it's been like this for years. _Sometimes_ when modelling exhaust headers I can get away with `Curves` > `Blend Curve` but anything with tight bends the user has to make a sketch with multiple arcs and tangent constraints to make a geometry pass model in my experience. Even with OCC 7.8.2dev it's no better.

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.3.33862 (Git)
Build type: Release
Branch: 021_Python_3_12_And_Other_Fixes
Hash: f9bffbb8b811a4f6213ec829c164ae09f7af0e50
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Curves 0.6.51
  * Assembly4 0.50.15
  * A2plus 0.4.68
```

ElementW, 2026-08-25:
Still relevant as of faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c and either OCC 7.9.3 or 8.1.0.dev1-3d097a0328

VM4Dim, 2026-09-02:
@kaiwas
Try uncheck `Data > Frenet` for `Sweep`
Verified in: `FreeCAD_weekly-2026.09.02-Linux-x86_64.AppImage`

kaiwas, 2026-09-02:
> Try uncheck `Data > Frenet` for `Sweep`

Yes. That helped. Thanks!
I can't figure out what "frenet" means, but if it's enabled by default and causing errors, I think it would be better to disable it by default.

What is this anyway? I can't find any usage examples.

VM4Dim, 2026-09-02:
https://wiki.freecad.org/Part_Sweep#Frenet

kaiwas, 2026-09-02:
> https://wiki.freecad.org/Part_Sweep#Frenet

Thnx!
Judging by the description, this sounds useful.
But should the round profile behave the way it does with the "Frenet" setting enabled? The current behavior doesn't seem appropriate to me.