FreeCAD/FreeCAD #17763: [Part] Thickness tool can't create a pipe from "Sweep + B-spline path" object

kaiwas, 2024-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If we created an object with the sweep tool - we cannot turn it into a through pipe with the thickness tool.

1. I create a pipe from a cylinder by selecting two faces (for testing).
2. The sweep object creates a container with a bottom (one face is selected, the thickness tool works.
3. The sweep object causes an error if we select two faces for the thickness tool.

There is a demonstration on the video.

https://github.com/user-attachments/assets/0694c345-93a2-4dce-9bfe-0e98a98f05ea

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39141 (Git) AppImage
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * addFC.backup1731047833.082574 0.7.4 (Disabled)
  * addFC 0.7.5
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-11:
Same as with lofts

chopsywa, 2024-11-23:
Lofts that fail in this v1 and dev versions work fine in 0.21.
[loft021.FCStd.zip](https://github.com/user-attachments/files/17879153/loft021.FCStd.zip)
[loft.FCStd.zip](https://github.com/user-attachments/files/17879154/loft.FCStd.zip)
Files are not zipped. Just extensions added.