FreeCAD/FreeCAD #17772: win: conda: too big qrc_resources file

looooo, 2024-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

We are running into this issue while trying to build freecad with qt6 

the error message is:
%SRC_DIR%\build\src\Gui\qrc_resource.cpp(1063568): fatal error C1060: compiler is out of heap space

A google search suggest to split the qrc-file. Maybe someone familiar with the qrc files can have a look at this and split the necessary files.

[1] https://github.com/conda-forge/freecad-feedstock/pull/122
[2] https://forum.qt.io/topic/116321/qrc_resources-cpp-fatal-error-c1060-compiler-is-out-of-heap-space/4

### Full version info

```shell
recent development snapshot.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-10:
@oursland FYI

h-vetinari, 2024-11-10:
This is a simple consequence of the build running out of memory and getting OOM-killed. I solved it by setting up a swap file.

The build still is heavy (both memory & disk usage); with the swap file in place, we manage to run out of disk space (on a drive that should have ~70GB available; minus all the stuff necessary for the build/host environment).

h-vetinari, 2024-11-10:
Actually the build fails while creating `FreeCAD.pdb`, which sounds like it's mostly debug-relevant? 🤔 Is there a way to suppress that creation?
```
[3636/5935] Linking CXX shared library bin\FreeCAD.pyd
FAILED: bin/FreeCAD.pyd src/Main/FreeCAD.lib 
C:\Windows\system32\cmd.exe /C "cd . && %BUILD_PREFIX%\Library\bin\cmake.exe -E vs_link_dll --intdir=src\Main\CMakeFiles\FreeCADMainPy.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100261~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100261~1.0\x64\mt.exe --manifests  -- C:\PROGRA~1\MICROS~2\2022\ENTERP~1\VC\Tools\MSVC\1441~1.341\bin\Hostx64\x64\link.exe /nologo src\Main\CMakeFiles\FreeCADMainPy.dir\FreeCADMainPy_autogen\mocs_compilation.cpp.obj src\Main\CMakeFiles\FreeCADMainPy.dir\MainPy.cpp.obj  /out:bin\FreeCAD.pyd /implib:src\Main\FreeCAD.lib /pdb:bin\FreeCAD.pdb /dll /version:0.0 /machine:x64 /NODEFAULTLIB /INCREMENTAL:NO /DEBUG  src\App\FreeCADApp.lib  src\Base\FreeCADBase.lib  %PREFIX%\Library\lib\xerces-c_3.lib  %PREFIX%\Library\lib\z.lib  MSVCRT.lib  MSVCPRT.lib  Rpcrt4.lib  version.lib  vcruntime.lib  ucrt.lib  concrt.lib  %PREFIX%\libs\python39.lib  %PREFIX%\Library\lib\boost_filesystem.lib  %PREFIX%\Library\lib\boost_program_options.lib  %PREFIX%\Library\lib\boost_regex.lib  %PREFIX%\Library\lib\boost_system.lib  %PREFIX%\Library\lib\boost_thread.lib  %PREFIX%\Library\lib\boost_date_time.lib  %PREFIX%\Library\lib\boost_chrono.lib  %PREFIX%\Library\lib\boost_atomic.lib  _deps\fmt-build\fmt.lib  %PREFIX%\Library\lib\Qt6Xml.lib  %PREFIX%\Library\lib\Qt6Core.lib  -lpthreads  mpr.lib  userenv.lib  kernel32.lib user32.lib gdi32.lib winspool.lib SHFolder.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib winmm.lib comsupp.lib Ws2_32.lib dbghelp.lib && cd ."
LINK: command "C:\PROGRA~1\MICROS~2\2022\ENTERP~1\VC\Tools\MSVC\1441~1.341\bin\Hostx64\x64\link.exe /nologo src\Main\CMakeFiles\FreeCADMainPy.dir\FreeCADMainPy_autogen\mocs_compilation.cpp.obj src\Main\CMakeFiles\FreeCADMainPy.dir\MainPy.cpp.obj /out:bin\FreeCAD.pyd /implib:src\Main\FreeCAD.lib /pdb:bin\FreeCAD.pdb /dll /version:0.0 /machine:x64 /NODEFAULTLIB /INCREMENTAL:NO /DEBUG src\App\FreeCADApp.lib src\Base\FreeCADBase.lib %PREFIX%\Library\lib\xerces-c_3.lib %PREFIX%\Library\lib\z.lib MSVCRT.lib MSVCPRT.lib Rpcrt4.lib version.lib vcruntime.lib ucrt.lib concrt.lib %PREFIX%\libs\python39.lib %PREFIX%\Library\lib\boost_filesystem.lib %PREFIX%\Library\lib\boost_program_options.lib %PREFIX%\Library\lib\boost_regex.lib %PREFIX%\Library\lib\boost_system.lib %PREFIX%\Library\lib\boost_thread.lib %PREFIX%\Library\lib\boost_date_time.lib %PREFIX%\Library\lib\boost_chrono.lib %PREFIX%\Library\lib\boost_atomic.lib _deps\fmt-build\fmt.lib %PREFIX%\Library\lib\Qt6Xml.lib %PREFIX%\Library\lib\Qt6Core.lib -lpthreads mpr.lib userenv.lib kernel32.lib user32.lib gdi32.lib winspool.lib SHFolder.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib winmm.lib comsupp.lib Ws2_32.lib dbghelp.lib /MANIFEST:EMBED,ID=2" failed (exit code 1201) with the following output:
LINK : warning LNK4044: unrecognized option '/lpthreads'; ignored

LINK : fatal error LNK1201: error writing to program database '%SRC_DIR%\build\bin\FreeCAD.pdb'; check for insufficient disk space, invalid path, or insufficient privilege

[3637/5935] Linking CXX executable bin\FreeCADCmd.exe
[3638/5935] Linking CXX executable bin\FreeCAD.exe
```

wwmayer, 2024-11-10:
For MSVC `/Zm150` is used. Is it also used for Conda builds? If yes, does it work to increase its value?

looooo, 2024-11-10:
for the conda-builds pch is disabled:
`-D FREECAD_USE_PCH:BOOL=OFF ^`

So I gues setting /Zm150 won't change anything. Is this correct?

looooo, 2024-11-10:
@wwmayer any ideas why the .pdb files are created, allthough we build a "release"?