FreeCAD/FreeCAD #17773: Part Disign: Make the active body identifier more visible in tree view. (Feature request)

kaiwas, 2024-11-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I don't work with part design very often, but even rare attempts have revealed the inconvenience of identifying the active body in the tree view.
Bold text is, of course, good, but it is not enough. Because fonts and screen resolutions can be different. As a result, the current version is very poorly identified.

Even if you make the text underlined, the active body will be more visible.

But, probably, some special identifier is required. For example, it could be a contrasting frame around the text.

The concept is in the picture.

[image]

PS/ I have the overlay disabled. If you enable it, the current version is perceived even worse. Bold text is very dependent on the design theme.
[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39141 (Git) AppImage
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * Curves 0.6.48
  * OpenTheme 2024.9.1
  * timber
  * PitchedRoof
  * CurvedShapes 1.0.10
  * FreeCAD_SketchArch
  * frame 0.1.1
  * lattice2 1.0.0
  * sheetmetal 0.4.26
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-11-11:
This is a style sheet issue. OOTB with the classic theme the active Body is pretty clear.
[image]

kaiwas, 2024-11-11:
> This is a style sheet issue. OOTB with the classic theme the active Body is pretty clear.

Yes. I mentioned that this parameter is heavily theme-dependent. Maybe there needs to be some way (Hardcode) to make the identification clear in 100% of cases?

mwganson, 2024-11-14:
It is possible to both underline and overline the active body.  Tools menu -> edit parameters -> baseapp -> preferences -> tree view -> TreeActive???? various options
[image]

You might need to create these boolean parameters if they do not already exist.  The color for the active object can be adjusted in edit menu - preferences -> display -> colors -> active container

kaiwas, 2024-11-14:
> You might need to create these boolean parameters if they do not already exist.

Excellent! Thank you!
So the issue is resolved even more simply than expected. Is it enough to make the settings built into the program by default?
Maybe it is worth making these parameters available by default?