FreeCAD/FreeCAD #17774: Measurment tool fails to work in assembly workbench

spiridonios, 2024-11-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have seen and read the related report " Measurement: tool doesn't work properly in the Integrated Assembly WB #14559 ",
which would describe quite well enough the issue i find, but since it is closed and has been asked to create a new post for related issues, i though it would be appropriate to create a new one. 

I have been trying to measure several types of distances with the new measurement tool. Sometimes it gives a result and sometimes it doesn't respond at all. Outside the assembly workbench i didn't have an issue with the tool. So even when a have a result in the assembly workspace, it is not what i ask for. A folder containing the measurements is not created at all and i have some readings like the ones in the attached image. I also included a zip file with the assembly file and all the affiliated fcstd files. Hope i did it properly.

Haven't post again here, and i hope i comply with the way i do it.. 
Great work generally with the new freecad version! I appreciate your hard work!

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Anything else?

[image]
[measurment.try.tar.gz](https://github.com/user-attachments/files/17690658/measurment.try.tar.gz)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-10:
Should work again in the latest weekly, please try: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

spiridonios, 2024-11-10:
I just downloaded the latest weekly. I still cannot take measurements between assembly objects. 
I can take measurements of the objects i created in the part workbench but it seems to be a problem with the room i have designed using the BIM workbench (in previous freecad version) and then making a fusion of the several BIM parts, using the fusion command of the part workbench in order to import this fusion into the assembly workbench and then to be able to design and add my furniture components. Could it be something in my workflow that cannot be computed with the measurement tool?

spiridonios, 2024-11-10:
when i try to measure some line of the "fused" room model or a point between the fused model and a newly imported part, i get a measurement indication at the point of origin, with a flag containing "0"

WandererFan, 2024-11-10:
If the objects being measured report themselves as belonging to the Assembly module, then Assembly needs a MeasureClient.  Part/App/MeasureClient only handles Part/PartDesign/Sketcher objects.

Without a measure client, there will be no geometry handler for measure to query.

luzpaz, 2024-11-23:
@hlorus https://github.com/FreeCAD/FreeCAD/issues/17774#issuecomment-2466752739 (any chance you can address this?)

WandererFan, 2024-12-20:
`Should we rename this to Assembly: Measurement function requires a measurement client in order for geometry handler for measure to query`

Done.