FreeCAD/FreeCAD #17775: Native freecad file won't open. error message

spiridonios, 2024-11-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have been working the assembly2.FCStd file since yesterday using the freecad 1.0.0RC2 version. Recently i dealt with another issue with the measurement tool and trying to find a solution a downloaded the weekly freecad version 39153-conda-Linux-x86_64-py311. At some point, after closing and trying to reopen my file i receive an error message (attached image) and now i see a file opened in freecad's treeview but actually nothing looks imported in the program. I also tried opening the file with the 1.0.0RC2 version, but i also received the same error message. I have no idea how it ended up like that. I don't think i did something inappropriate. At this point the easiest thing for me would be to try and assemble from the beginning. I write this report though in case it can help with the development of the software. 

here is what the report view says:

12:02:25  Fatal Error at file /home/spiridonious/Downloads/krevati_peristera/assembly2.FCStd, line 273, char 27
12:02:25  Invalid Document.xml: unexpected end of input
12:02:25  Fatal Error at file GuiDocument.xml, line 1, char 1
12:02:25  Exception message is: 
12:02:25  invalid document structure
12:02:25  Reading failed from embedded file: GuiDocument.xml (0 bytes, 2 bytes compressed)
12:06:42  Fatal Error at file /home/spiridonious/Downloads/krevati_peristera/assembly2.FCStd, line 273, char 27
12:06:42  Invalid Document.xml: unexpected end of input
12:06:42  Fatal Error at file GuiDocument.xml, line 1, char 1
12:06:42  Exception message is: 
12:06:42  invalid document structure
12:06:42  Reading failed from embedded file: GuiDocument.xml (0 bytes, 2 bytes compressed)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (Enlightenment/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.39153 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Anything else?

[image]
[thefolderiwork_with.tar.gz](https://github.com/user-attachments/files/17690990/thefolderiwork_with.tar.gz)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-11-10:
The file assembly2.FCStd is indeed corrupted. Its Document.xml is not complete and GuiDocument.xml is completely empty. The backup file assembly2.20241110-114022.FCBak is corrupted, too.

In order to continue you can rename assembly2.20241110-113518.FCBak to assembly2.FCStd.

But the question how it happened to break the XML files I cannot answer either.

spiridonios, 2024-11-10:
thank you for the renaming hint wwmayer. It was helpful!

danielbot, 2024-11-12:
"Me too". Confirmed: hit the same bug on weekly-builds-39153. I can't say much about the sequence leading up to it, but again both the file and today's backups were corrupted, while yesterday's backup produced by weekly-builds-39141 was usable. No edits were done to the corrupted document between yesterday and today, or for several days before that.

wwmayer, 2024-11-12:
The commit for build 39153 is 3a7c8ed and for build 39141 presumably 4bd80bdc. When looking at the code changes at each commit in between I can't see a possible regression causing the reported problems.

nordtechtiger, 2024-12-05:
Just happened to me too, exactly the same error. Thankfully the backup worked - I still lost some work but it's better than having to start over from scratch again.

danielbot, 2024-12-06:
Re nordtechtiger... similar experience for me. I also see what looks like namespace corruption, where objects can't find their referenced varsets, even though I can see them right there, intact, in the tree. After renaming the unfindable object, the expression editor is able to parse the expression, using the new name. Then I can rename the object back to the original name and the reference stays intact... for a while. Recompute then breaks other references randomly. I think this may be related to what we're seeing here. I will try to reduce the fairly complex document this happens in and post the (fragile and partly broken) document(s) here.

Also, deleting some hidden copyonwrite groups (desperation measure to repair a badly broken model using tracking links) also deleted an underlying sketch referenced by several non-copyonwrite objects. This smells like a reference count bug to me, or in other words object lifetime. I would suggest adding some paranoia code to the reference count decs to complain loudly if they ever go negative, or better, if they ever go to zero while still referenced. (The latter being a modest challenge, but not that bad... just scan every object in the document to see if any reference still the object now being deleted when reference count _apparently_ went to zero.)

danielbot, 2024-12-06:
Another thing I am seeing is link properties I added as custom properties being randomly deleted from objects. Not sure which operations cause this. Maybe the automatic global recomputes that happen from time to time, triggered by circumstances that aren't clear.

nordtechtiger, 2024-12-07:
Some more info: It has happened a couple of times and I'm able to reproduce it. Ever since my files started getting corrupted, I started watching the file size every time I saved since it tends to drops by ~75%.

This is the workflow that triggers the bug for me:
1. Import a part into an assembly
2. CTRL+Z to remove the part
3. CTRL+S to save

The file is now corrupted, and will stay that way UNLESS you use CTRL+SHIFT+Z to redo everything until you're back to the newest version. If you save at this stage, the file will recover.

I'm still not sure whether the parts & assemblies are crucial for reproducing this bug, but I haven't had more time to narrow it down. Pretty sure it's CTRL+Z that's the culprit.

I can see this tying into the issues you are experiencing, Daniel. It for sure smells like some dangling pointers/references or something...

PaddleStroke, 2024-12-13:
I have tried to reproduce but can't.

@nordtechtiger can you please detail a bit?
1 Are you using latest weekly?
2 - By import you mean 'insert part'? By Part do you mean a App::Part container or any solid?
3 - you ctrl-Z after finishing the tool?

If you can provide a reproduction file and precise reproduction steps please?

I have a feeling that it may be related to https://github.com/FreeCAD/FreeCAD/pull/16671 but not sure at all yet.

nordtechtiger, 2024-12-26:
My apologies for not responding sooner - I just discovered those steps while working and just thought I'd post them. Hoping I remembered all the details correctly...

1. I'm using the latest package for Arch.
```
$ freecad --version
FreeCAD 1.0.0 Revision: 39109 (Git)
```

2. I double clicked an assembly, selected "insert component", and selected a body from my document.

3. After inserting it into the assembly, I hit Ctrl-Z after the part was inserted but while the assembly was still active.

I will keep testing and let you know if I find anything definitive.

miscounted, 2025-01-14:
Me too, 39109 GIT, debian 12 x86_64
Every single entry in main file and backup is 0 bytes with the exception of Document.xml

It went so fast, I do not know what happened but I'm doubtful that I used ctrl+z. Maybe it has something to do with the fact, that my peojects directory got only 234,1MiB free. Maybe the deflated file ran out of space, before the freecad's compression (level 7).

PaddleStroke, 2025-01-14:
> After inserting it into the assembly, I hit Ctrl-Z after the part was inserted but while the assembly was still active.

@nordtechtiger can you please specific if you used Ctrl-Z after closing the 'Insert component' task? Or did you use Ctrl-Z while the task dialog was still there?

nordtechtiger, 2025-01-14:
@PaddleStroke Unfortunately it's been a while since I ran into the issue, and it's very possible that the issue only showed up in the first place because of something I had done earlier in the project. I have since downloaded the AppImage of the latest in-progress version and all of my issues have been solved. I'm unsure whether this is because of the version differences or whether it's because of the distribution method (AppImage vs pacman), but I wouldn't be surprised if it's a library mismatch somewhere perhaps?

VynDragon, 2025-02-27:
> That's the same version I used when my files got corrupted. How did you install FreeCAD? [@VynDragon](https://github.com/VynDragon)

It's the current version on archlinux repositories.

nordtechtiger, 2025-03-01:
Similar experience here.