FreeCAD/FreeCAD #17776: Toponaming FreeCad 1.0 latest build

jimsybeldon, 2024-11-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Latest build is being used:

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: https://github.com/FreeCAD/FreeCAD/commit/3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods:

A2plus 0.4.61
Assembly3 0.12.0
Assembly4 0.50.6
Help 1.0.3

Regarding the TNP issue:
[Model for TNP testing.zip](https://github.com/user-attachments/files/17683697/Model.for.TNP.testing.zip)

This is the assembly before I modify an existing part
[image]

Model provide is a stable, simple assembly for FC1. but is 
It is provided purely to establish that a TNP issue will effect this model. 
On my local machine I  added a hole feature to the "ChimneyShipLadSide" part.  
A naming issue is seen is the report view:

14:03:00 TopoShapeExpansion.cpp(2019): Invalid shape name ?Face3
14:03:00 Assembly: Invalid shape name ?Face3
14:03:00 TopoShapeExpansion.cpp(2019): Invalid shape name ?Face3
14:03:00 Assembly: Invalid shape name ?Face3
14:03:05 TopoShapeExpansion.cpp(2019): Invalid shape name ?Face3
14:03:05 Assembly: Invalid shape name ?Face3
14:04:06 TopoShapeExpansion.cpp(2019): Invalid shape name ?Face3
14:04:06 Assembly: Invalid shape name ?Face3

This is the model after the part mod:
[image]
08:28:31  <PropertyLinks> PropertyLinks.cpp(429): FirePlaceSketchHolder#Joint137.Reference1 auto change element reference FirePlaceSketchHolder#ChimneyShipLadSide003 ChimneyShipLadSide003.;Vertex3;:H1100,V.Vertex3 -> ChimneyShipLadSide003.;#334;:Hff2,V;:H1100,V.Vertex3
08:28:31  <PropertyLinks> PropertyLinks.cpp(429): FirePlaceSketchHolder#Joint137.Reference2 auto change element reference FirePlaceSketchHolder#ChimneyShipLadSide ChimneyShipLadSide.;Vertex11;:H1100,V.Vertex11 -> ChimneyShipLadSide.;#33d;:Hff2,V;:H1100,V.Vertex15
08:28:31  <Exception> TopoShapeExpansion.cpp(2019): Invalid shape name ?Face4
08:28:31  Assembly: Invalid shape name ?Face4

I have been able to replicate this consistently.

I realize there are other TopoNaming issues out there.

TNP

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.61
  * Assembly3 0.12.0
  * Assembly4 0.50.6
  * Help 1.0.3
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jimsybeldon, 2024-11-10:
Toponaming issues that are open that will effect Assy WB

https://github.com/FreeCAD/FreeCAD/issues/17412
https://github.com/FreeCAD/FreeCAD/issues/17266
https://github.com/FreeCAD/FreeCAD/pull/16256
https://github.com/FreeCAD/FreeCAD/issues/15486
https://github.com/FreeCAD/FreeCAD/issues/14720
https://github.com/FreeCAD/FreeCAD/issues/14643

jimsybeldon, 2024-11-14:
Hey anyone,

Is this the situation that puts toponaming and assembly work bench issues on extended hold?
https://ondsel.com/blog/goodbye/t
Seems it is.

Dang.....

jimsybeldon, 2024-11-14:
opps-

https://ondsel.com/blog/goodbye/

They are shut down.....

oursland, 2024-11-14:
That's part of it, but another part was that a major contributor to the Toponaming work, Brad McLean, suffered a fatal heart attack in October.  [The version 1.0 is dedicated in his honor.](https://blog.freecad.org/2024/10/08/rip-bradley-bgbsww-mclean/comment-page-1/)

The TNP is a very challenging problem.  Others are picking up where he left off, but it will take time to develop the same knowledge and skill that he had.

drwho495, 2025-07-27:
this was not an issue with assembly, its a general issue with applying names
might be fixed from https://github.com/FreeCAD/FreeCAD/pull/22767

i will test soon

luzpaz, 2025-08-09:
>  also adding features causes errors typically, its a known issue that will be fixed in a few months  

Is there an open ticket for that we can link this ticket to ?