FreeCAD/FreeCAD #17780: Bim: ability to move carpentry when moving a line of a wall

gianlucafiore, 2024-11-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be good to always keep the joinery aligned and adhered to the walls, when one creates a series of walls with a skectcher plan

[image]
[image]
[image]
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/Fusion
Installed mods: 
  * dodo 1.0.1
  * fasteners 0.5.29
  * FEMbyGEN 2.2.0
  * Nodes 0.1.36
  * SearchBar 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-11-11:
Windows and doors assign their coordinates once, relative to the wall origin. When changing the position of the wall edge, the wall origin does not change, so the windows and doors remain in the same place.
Solution options:
1) use expressions to calculate the coordinates of the products.
2) Use the SketchArch addon  (but it also has its drawbacks)
https://github.com/paullee0/FreeCAD_SketchArch

gianlucafiore, 2024-11-11:
Good morning, excellent, thank you for your answer and for said complement, I did not know the same one to install, I am going to give it a good view, surely it will be very useful. Best regards

paullee0, 2024-11-11:
>     2. Use the SketchArch addon  (but it also has its drawbacks)
>        https://github.com/paullee0/FreeCAD_SketchArch
> 
> 
> But yes. I agree that this is a huge problem. And workarounds are often very labor-intensive.
> 
> We discussed with Paulee for a long time possible solutions in his add-on. Perhaps the situation will change someday. [paullee0/FreeCAD_SketchArch#18](https://github.com/paullee0/FreeCAD_SketchArch/issues/18)

Thanks @kaiwas for bringing this up.

Documentation is lacking, the original forum discussion and several Youtube videos maybe found below -

- [ SketchArch Addon - Parametric Windows/Doors/Equipment Placement, TNP-Tolerant](https://forum.freecad.org/viewtopic.php?style=2&p=600105#p600105)

kaiwas, 2024-11-11:
In fact, this is a sore subject. Windows and doors in principle should not exist separately from the wall in which they are built. The only exception is windows and doors as an autonomous object without a host.

MrFoldAnyTen, 2024-11-11:
This is my first git hub comment as I'm interested, I'm making my own joinery products, but the location of a third party  object needs to be made relative to something in the model if you have a datebase or an expression involved in the position of the wall then the position of the door can workfrom that, I'm making windows and doors with Python script so am interested in how the user would like to reference their geometry.

MrFoldAnyTen, 2024-11-11:
Can you just go to the assembly workbench and constrain the frame with offsets as necessary tothe edge of 1 wall the face of a wall and the floor or something like that?

paullee0, 2024-11-12:
> This is my first git hub comment as I'm interested, I'm making my own joinery products, but the location of a third party object needs to be made relative to something in the model if you have a [...]

Could you make a small demo of your development?

My add-on works by adding Sketch with additional features  (Sketch becomes SketchObjectPython which I call it ArchSketch).  E.g. ArchWindow is attached an ArchWall making referencing to its Base Sketch object, a particular edge user selects.   And it follows whenever user edit the Sketch.  This is a simple outline of the algorithm.

Related TNP issues on BIM for reference
https://github.com/FreeCAD/FreeCAD/issues/14927

kaiwas, 2024-11-12:
Hardly anyone uses an assembly workbench for architectural design.
Using the standard program methods, you can bind windows and doors to a wall using expressions. This is labor-intensive, but it works if you are careful. A small demonstration. (You can also bind the length of one wall to another wall.

https://github.com/user-attachments/assets/5052f661-75a4-4dee-95ce-bd2584458357

This is all very tedious, because you constantly need to think about what to bind to and what values ​​​​should be used in the calculations.

paullee0, 2024-11-22:
BTW, this is already possible as shown in a serious of videos posted on Youtube earlier in 2020 
(E.g. https://www.youtube.com/watch?v=7amrcUmd1YA)

First one is here, see 1:50 particularly :- 

https://github.com/user-attachments/assets/09e775b2-2dfc-49ac-99d0-8193fb2398d0

paullee0, 2025-04-18:
Seems several issues created on similar feature request.

Check this PR / Forum Discussion:
- https://github.com/FreeCAD/FreeCAD/pull/20872
- https://forum.freecad.org/viewtopic.php?p=822493#p822493

https://github.com/user-attachments/assets/164b368e-100c-4a66-914a-2462f0a7a1c5