FreeCAD/FreeCAD #17783: Assembly: distance joint mulfunction between subassemblies

spiridonios, 2024-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I see several post about problem with the assembly joints, but didn't locate something that i understand as the same so i post here. 
Continuing designing my model, i have several freecad crashes (it closes unexpectedly with no error messages to read), but i reopen the software, continue my work and save regularly. At this point i have an issue trying to create a distance joint between two subassemblies in my model. As seen in the image, the one circles is a different sub assembly than the one on the right. I try to make a distance joint between their two surfaces. At first the circles model was moving directly into the refereed surface of the right assembly, but i was able to move it on the X axis back where i wanted it. After i opened the joint to give a 20mm distance between the two, but it looks like the software cannot translate this well. At first it moved 20mm but shortly crashed and after opening freecad again and doing the same commands i cannot have the distance appear (at some point the circled part was moving only from one side on Y axis, meaning it was not being kept parallel, and that without using any rotation commands). Now when i try to grab and move the circled part i receive the message shown in the image ("message"). 
Trying to grab any part and move it again i receive the "new message" picture window open and the model being a little be messed up ("modelupsidedown") .

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (Enlightenment/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.39153 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

spiridonios, 2024-11-11:
[mworkingfolder.tar.gz](https://github.com/user-attachments/files/17698490/mworkingfolder.tar.gz)
here i attach my working folder as it now

spiridonios, 2024-11-16:
I ve been trying to open the laptop where i created and keep my file yesterday but for some reason now it doesn't boot.. So now i have to sort this out as well. So for now i am not able to try to create and send the simplified version you asked for PaddleStroke. sorry..

luzpaz, 2025-03-15:
@spiridonios any progress ?

spiridonios, 2025-03-15:
It's been a while since i did any project with freecad. For this project i did try and remake the assemblies. In the end i used a macro script (explained here: [](https://github.com/FreeCAD/FreeCAD/issues/18114)) because of another assembly workbench related issue and i was generally able to come up with a useful model. I m not sure how much this script could have helped to this issue as well though..

Jocarnail, 2025-06-28:
I have the same problem: here is the minimal file I can generate. Shamefully... it's not very minimal

[CrashTest_assembly.tar.gz](https://github.com/user-attachments/files/20961722/CrashTest_assembly.tar.gz)

When the uppermost object is clicked the program crash. 

I tried to make a more minimal version but was not able to reproduce the issue: neither in a new project or in the same one.
If I remove the bottom object from the assembly (the one below in the 3D view, not the assembly tree) the distance joint works without issues.

The object that causes the crash can still be manipulated from the tree without crashing. I was able to transform it without issues. Moving the transformation widget caused no issues.

There also seems to be a problem saving: when trying to save with the keyboard shortcut after creating the joint nothing happens. However, when prompted to save when closing the window the file was correctly saved.

PaddleStroke, 2025-07-18:
So there are 2 issue reports here.
1 - the original report. Which I am going to summarize for reference : 
Open assembly3
create a distance joint between Cube and Assembly005 flat faces.

I can reproduce the error messages : 
Error messages printed.
Distance joint misbehaving.

I think this is very likely an overconstraint assembly but will investigate.

@Jocarnail I don't think your issue is the same as the OP here does not have a crash.
I checked your file but many files are missing so it cannot be opened. If you want please open your own issue with minimal file (complete) and reproduction steps.