FreeCAD/FreeCAD #17784: Part Design offers to create a body for a sketch, then complains the sketch doesn't belong to it

Stokestack, 2024-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you try to perform an operation on a sketch that requires a body, FreeCAD will offer to create a body for it. If you accept that offer, FreeCAD immediately complains that they sketch doesn't belong to that body.

You can demonstrate this by starting a new file, making a sketch, and trying to pad it. Screen grab attached.

https://github.com/user-attachments/assets/aa7675ac-8edc-465b-96db-768b7a05198c

### Full version info

```shell
OS: macOS 15.1
Architecture: arm64
Version: 1.1.0dev.39153 (Git)
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: Dark-modern.qss/Dark behave/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-11:
Cannot reproduce.
- Without a body, a new body is created and activated
- With one inactive body, the body is activated ans sketch placed inside
- With more than one body but no active body, if I choose to create a new body, the new body is created, activated and the sketch is placed inside. No issues or errors.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.27.14
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.3
  * Silk 0.1.5

```

luzpaz, 2024-11-11:
@maxwxyz Does it matter you're using r39141 and op is using r39153 ?

maxwxyz, 2024-11-11:
@luzpaz just tested on newest one, still no issue.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39153 (Git)
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.27.14
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.3
  * Silk 0.1.5

```

Stokestack, 2024-11-11:
Well, I'm on the Mac version. Seems weird that this would differ by platform, though...

maxwxyz, 2024-11-11:
@JohnOCFII maybe you can give this a try?

Stokestack, 2024-11-11:
OK, here are two ways to reproduce this.

A:
1. Launch FreeCAD.
2. Select the Sketch workbench and create a new document.
3. Create a new sketch.
4. Switch to Part Workbench and try to pad the sketch.

B.
1. Create a body in Part Workbench and add a sketch to it.
2. Drag the sketch to the top of the model tree.
3. Try to pad the sketch.

luzpaz, 2024-11-11:
>  Seems weird that this would differ by platform, though...

If you hang on any  open source multi-platorm supporting project bugtracker you'll see quickly there are all sorts of nuances between different platforms and even between different Linux distributions. So not surprising at all.

Stokestack, 2024-11-11:
That comment was based on the "workflow" nature of this particular issue. Anyway, I sussed out the exact repro steps so I'm guessing this will happen on other platforms as well.

maxwxyz, 2024-11-11:
I think you've messed up the WBs. You've mentioned PartDesign in the original issue and now Part WB.
If you want to work in PartDesign, you can create the Sketch from the PartDesign WB, no need to manually switch to the Sketcher WB.
I was able to reproduce the issue when starting in the Sketcher WB and then trying to pad it in PartDesign. IMO this is no issue, as even when the Sketch would be moved into the body, the sketch has no attachment which could lead to more issues later.

closed 2024-11-11 by maxwxyz

Stokestack, 2024-11-11:
That's true, I meant Part Design.

maxwxyz, 2024-11-11:
Yes but the Sketch is is still lacking an attachment.

Stokestack, 2024-11-11:
An attachment to what?