FreeCAD/FreeCAD #17785: When opening a sketch, parent bodies' visibility is re-enabled, obscuring your view of the sketch

Stokestack, 2024-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When building a part in Part Design and trying to edit a sketch, the visibility of the parent (or previous body or whatever it might be called) it is re-enabled, which blocks your view of the sketch you're trying to edit. You then have to go back to the model and re-disable the other body's visibility. Screen grab attached.

https://github.com/user-attachments/assets/45f17ad2-6797-4b9a-a2f4-cf8b8c6dd431

### Full version info

```shell
OS: macOS 15.1
Architecture: arm64
Version: 1.1.0dev.39153 (Git)
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: Dark-modern.qss/Dark behave/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-11:
That is not a bug but the sketch is inside and if you hide the parent the sketch is hidden too. You can activate the "View section" command to see the sketch.

closed 2024-11-11 by maxwxyz

Stokestack, 2024-11-11:
That is incorrect. Please watch the video.

https://github.com/user-attachments/assets/0262c814-de0e-48bb-8ca7-efda23e4fa10