FreeCAD/FreeCAD #17786: Freeecad 1.0 rc2 cannot save to network drives on Windows 10

jkalousek, 2024-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While saving document on Windows 10 save dialog will not show network drives under FC 1.0 rc2:
[image]

Same operation under 0.21.2 works as expected and I'm able to save without problem:
[image]

For me this is Quite bid deal as I'm not using local storage on PC and everything goes to SMB server.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Czech/Czech Republic (cs_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * 3DfindIT 1.2.0
  * Assembly4 0.50.15
  * dxf-library
  * freecad.gears 1.0.0
  * Gridfinity 0.6.0
  * parts_library
  * sheetmetal 0.2.60
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-11-11:
Does the latest _FreeCAD_ executable has Internet/LAN access? I have encountered this before, the stable version had access to network, while the latest downloaded version of _FreeCAD_ did not, _Windows Firewall_ forbade it.

----

If you type directly the network folder, like `\\192.168.0.4\public_diskF`, does it open the directory?

If you press right click, do you have item to _Add network shared folder_? Does it work?

maxwxyz, 2024-11-11:
Cannot reproduce on:
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.27.14
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.3
  * Silk 0.1.5

```

jkalousek, 2024-11-11:
Do not know what changed but after restarting FC 1.0 rc2 for 3 times network drives appeared as expected. I saw in the past software that deliberately didn't want to store files on network drive and did hide mapped network drives so I assumed that it may be same.

closed 2024-11-11 by jkalousek