FreeCAD/FreeCAD #17787: FreeCAD 1.0 RC2/weekly freeze when opening AddonManager

ad-on-is, 2024-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening the AddonManager and hitting "OK", freecad just freezes.

```
FreeCAD 1.1.0, Libs: 1.1.0devR39153 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
QApplication: invalid style override 'kvantum' passed, ignoring it.
	Available styles: Windows, Fusion
Migrating Start Workbench to Start command... done.
An error occurred updating macros from GitHub, trying clean checkout...:
Git returned a non-zero exit status: 128
Called with: /usr/bin/git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD-macros.git /home/adonis/.cache/FreeCAD/Cache/AddonManager/Macros

Returned stderr:
Cloning into '/home/adonis/.cache/FreeCAD/Cache/AddonManager/Macros'...
fatal: unable to access 'https://github.com/FreeCAD/FreeCAD-macros.git/': .netrc parser error

/home/adonis/.cache/FreeCAD/Cache/AddonManager/Macros
Attempting to do a clean checkout...
Traceback (most recent call last):
  File "/tmp/.mount_freecalPiagM/usr/Mod/AddonManager/addonmanager_git.py", line 455, in _synchronous_call_git
    proc = utils.run_interruptable_subprocess(final_args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_freecalPiagM/usr/Mod/AddonManager/addonmanager_utilities.py", line 429, in run_interruptable_subprocess
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['/usr/bin/git', 'clone', '--recurse-submodules', 'https://github.com/FreeCAD/FreeCAD-macros.git', '/home/adonis/.cache/FreeCAD/Cache/AddonManager/Macros']' returned non-zero exit status 128.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/tmp/.mount_freecalPiagM/usr/Mod/AddonManager/addonmanager_workers_startup.py", line 339, in _update_local_git_repo
    self.git_manager.clone(
  File "/tmp/.mount_freecalPiagM/usr/Mod/AddonManager/addonmanager_git.py", line 90, in clone
    self._synchronous_call_git(final_args)
  File "/tmp/.mount_freecalPiagM/usr/Mod/AddonManager/addonmanager_git.py", line 457, in _synchronous_call_git
    raise GitFailed(
addonmanager_git.GitFailed: Git returned a non-zero exit status: 128
Called with: /usr/bin/git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD-macros.git /home/adonis/.cache/FreeCAD/Cache/AddonManager/Macros

Returned stderr:
Cloning into '/home/adonis/.cache/FreeCAD/Cache/AddonManager/Macros'...
fatal: unable to access 'https://github.com/FreeCAD/FreeCAD-macros.git/': .netrc parser error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/tmp/.mount_freecalPiagM/usr/Mod/AddonManager/addonmanager_workers_startup.py", line 92, in run
    self._retrieve_macros_from_git()
  File "/tmp/.mount_freecalPiagM/usr/Mod/AddonManager/addonmanager_workers_startup.py", line 288, in _retrieve_macros_from_git
    update_succeeded = self._update_local_git_repo()
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_freecalPiagM/usr/Mod/AddonManager/addonmanager_workers_startup.py", line 356, in _update_local_git_repo
    os.chdir(
FileNotFoundError: [Errno 2] No such file or directory: '/home/adonis/.cache/FreeCAD/Cache/AddonManager/Macros/..'

```

### Full version info

```shell
FreeCAD 1.1.0, Libs: 1.1.0devR39153 (Git)
FreeCAD 1.0.0 Revision: 38806 (Git)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

huguesdpdn-aerospace, 2024-11-11:
Hello @ad-on-is , I have a Linux, running the latest weekly build (`39153` as yours) and have no issue with the AddonManager.

The other issue is that I don't have a "OK" button. Can you please share a screenshot where you click on the OK ?

For info, my addon manager path is `/home/xxxxxxxxxx/.cache/FreeCAD/Cache/AddonManager/Macros` but no `Macros.backupxxxxxxx.xxxx`

ad-on-is, 2024-11-11:
@huguesdpdn-aerospace I already removed these folders. 

It all started when I installed RC2 today and experienced a toolbar UI issue (unrelated). Since I tried FC on and off, but used Ondsel, I thought it might be a good idea to do a clean install and deleted said folders.

When I went ahead to install some addons (unrelated to the current theme) FC RC2 froze. Then I installed the weekly, which also froze. I then tried RC1 and it works.

Can you run it from the command line and see if you get the error when AM tries connecting to GitHub?

FWIW:
- openSUSE Tumbleweed
- WM Hyprland (wayland)
- FC installed using appman

ad-on-is, 2024-11-11:
@huguesdpdn-aerospace 
```
OS: openSUSE Tumbleweed (Hyprland/hyprland/xcb)
Architecture: x86_64
Version: 1.1.0dev.39153 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

Here's a video of proof moving the folders.

https://github.com/user-attachments/assets/ad53a71e-30a5-4606-8cc0-ff405f28165e

ad-on-is, 2024-11-11:
This is why I also wrote about the ~/.cache/FreeCAD folder only. Previously, I tried renaming the folders one-by-one to see which one causes the issue, to narrow it down for you guys.

huguesdpdn-aerospace, 2024-11-11:
Very nice video, thanks and thank you again. For info, it also take me like 3 seconds for to let FreeCAD download all the addons list for the first time. Because the video is ended a bit early but I don't know if, if the video was longer, the addon manager was succeeding to get the list as I. 

OK so now I may have a better idea of the issue, here is what I suspect: is a invalid git configuration with your `.netrc` file (so Option 3)

Can you please send us the full output of
```
cd ~
mkdir test
cd test
git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD-macros.git ~/test

mkdir -p /home/adonis/.cache/FreeCAD/Cache/AddonManager/Macro
/usr/bin/git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD-macros.git /home/adonis/.cache/FreeCAD/Cache/AddonManager/Macros
```
This is to reproduce, by hand, GIT commands executed in FreeCAD. Let see the output.

Can you also copy/paste the output for:
```
echo "==== Files with GIT ====" ; find ~ -type f -iname "*git*"

echo "==== .netrc file START ====" ; cat .netrc ; echo "==== .netrc file END ===="

echo "==== .ssh folder==== ; if [ ! -d ~/.ssh ]; then echo "No .ssh folder" ; else ls -la ~/.ssh/* ; fi

echo "==== git config ==== ; git config --list

echo "==== git config ORIGIN ==== ; git config --list --show-origin
```

# What I see
If GIT return a `128` code , it means you have an issue and not FreeCAD. And actually the GIT behavior changed between RC1 et RC2 with some kind of cache, so I am not surprised if it works with RC1 and not RC2 or weekly.

ad-on-is, 2024-11-11:
@huguesdpdn-aerospace sorry for cutting off the video that early... I thought it was already clear by then that it froze.

huguesdpdn-aerospace, 2024-11-11:
No problem, since I guess the issue is still here. I am pretty sure it is a GIT issue. Copy/paste the above output commands when time permits :)

ad-on-is, 2024-11-11:
I'm so sorry for the trouble ... and very thankful to you for taking the time and patience with me. It never occurred to me that the issue might be related to something else. I will report back, once I fix the git issue.

huguesdpdn-aerospace, 2024-11-11:
There you go :) So either you know how to fix your `.netrc` file, either you can check the official documentation:
https://www.gnu.org/software/inetutils/manual/html_node/The-_002enetrc-file.html

@maxwxyz : you can close this issue

@ad-on-is : Have a nice day :)

ad-on-is, 2024-11-11:
I never touched that file recently, it had some old entries from a previous client I worked with.
However... clearing the .netrc file solves the issue... and weekly works now without issues.

However, this might still be kept in consideration, to not just freeze if there's an underlying error with git, or the network or anything else.

@huguesdpdn-aerospace Thank you, have a nice day as well.

closed 2024-11-11 by ad-on-is