FreeCAD/FreeCAD #17788: Build error with docker image

Baptou88, 2024-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello everyone
I have been looking for an open source project that could allow me to develop features for a long time, and I just discovered this incredible project!
I am looking to improve the CAM workshop to be able to use it in business. Indeed, I am currently using EasyMill (Very pleasant to use, but very long updates to wait, and no possibility of adding mods...)

I come to you today because I cannot compile this software from source, having followed [this tutorial](https://wiki.freecad.org/Compile_on_Docker#Build_FreeCAD)

when I try to compile using this command: `/root/build_script.sh`
I get this message:
`CMake Error at cMake/FreeCAD_Helpers/CompilerChecksAndSetups.cmake:39 (message):
  FreeCAD 0.20 and later requires C++17.  G++ must be 7.3 or later, the used
  version is 6.3.0`

Cordially Bapt'

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-11:
@oursland FYI

luzpaz, 2024-11-11:
@Baptou88 0.21.2 is pretty outdated. We've just been stalled on the 1.0 release cycle for a while. We're in RC2 (2nd release candidate). Could you instead use RC2 as the docker image ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2 ?

Baptou88, 2024-11-11:
Thank you for your answer!
The information I put on top were just to show that I was using Windows 11.

When I clone this repo I've used this command: `git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD.git`

oursland, 2024-11-11:
The Docker container images used in the wiki are 5 years old.  I suspect those instructions are out of date.

I do not know where the most recent build instructions using Docker are hosted.  There are `Dockerfiles` checked into source in this directory: https://github.com/FreeCAD/FreeCAD/tree/main/tools/build/Docker

Baptou88, 2024-11-11:
Thanks for your reply!
I've check this folder but no file seems to be for windows...

luzpaz, 2024-11-11:
> The Docker container images used in the wiki are 5 years old. I suspect those instructions are out of date.

ITC, we should open up a new ticket to update Docker documentation in the wiki

Baptou88, 2024-11-12:
Therefore what is the right solution for build this project? Pass under Linux (what distribution)?

Baptou88, 2024-11-13:
I've found this tutorial: https://wiki.freecad.org/Compile_on_Windows

luzpaz, 2024-12-02:
Anyone willing to update the docs ?

JohnOCFII, 2025-10-17:
> On https://wiki.freecad.org/Compile_on_Linux, under Ubuntu, the needed package list for Qt6 is missing is missing python3-yaml (needed to test the CAM workbench). The Qt5 package list is missing that and python3-pyside2.qtnetwork. I haven't looked at the other build environments.

Just FYI -- much of the Wiki build info is a little stale.  More recent build info (especially the Pixi build instructions) can be found in the Developer's Handbook: https://freecad.github.io/DevelopersHandbook/