FreeCAD/FreeCAD #17791: Creation of two separate solid bodies failed

Roberto8303, 2024-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

by setting allow more solids in the body in partdesign by creating two rectangles and then extruding only one appears
[image]
[image]
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39153 (Git)
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * FreeCAD-themes 2024.7.24
  * Glass
  * IconThemes
  * OpenTheme 2024.9.1
  * PieMenu 1.8.1
  * Render 2024.9.25
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-11-11:
Please check if the "Allow Compound" property of the Body is set to true.

Roberto8303, 2024-11-11:
was set to false
 it works thanks

closed 2024-11-11 by Roberto8303