FreeCAD/FreeCAD #17792: Kubuntu 22.04 FC0.212 Part_Disign Hole

duinzicht8, 2024-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Part Design_Hole crashes by making Hole.

lesson https://www.youtube.com/watch?v=zfmsIz2wMGw

1. Open FreeCAD
2. Press the "create new document" button
3. Switch to the PartDesign workbench
4. Press the "Create body" button
5. Make sketch
6. Make cicle
7. make multi line
8. 2 x sketcher constrainpointonproject
9. scetcherconstaindistance
10. close sketcher
11. Partdesign revolution
12. Open sketcher
13. Make cicle
14. Scetch scetch view
15. 2x scetcherconstaindistance
16. scetcherconstainCircleDia
17. close
18. PartDesign_Hole
19. FreeCAD crashes

### Full version info

```shell
PRETTY_NAME="Ubuntu 22.04.5 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.5 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

[code]
OS: Ubuntu 22.04.5 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * woodworking 0.21.2.33771
  * btl 0.9.9
[/code]
```

### Subproject(s) affected?

None

### Anything else?

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7fd2bee73520]
#1  0x7fd171b6c78c in PartDesign::Hole::findHoles(TopoDS_Shape const&, TopoDS_Shape const&) const from /usr/lib/freecad-python3/lib/_PartDesign.so+0x22c
#2  0x7fd171b72f5c in PartDesign::Hole::execute() from /usr/lib/freecad-python3/lib/_PartDesign.so+0x114c
#3  0x7fd2c0f6f56a in App::DocumentObject::recompute() from /usr/lib/freecad-python3/lib/libFreeCADApp.so+0x3a
#4  0x7fd17175a22d in Part::Feature::recompute() from /usr/lib/freecad-python3/lib/Part.so+0xd
#5  0x7fd2c0f419cd in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib/freecad-python3/lib/libFreeCADApp.so+0x31d
#6  0x7fd2c0f3dd94 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /usr/lib/freecad-python3/lib/libFreeCADApp.so+0x584
#7  0x7fd2c0fbdb11 in App::DocumentPy::recompute(_object*) from /usr/lib/freecad-python3/lib/libFreeCADApp.so+0x2f1
#8  0x7fd2c0fb460c in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /usr/lib/freecad-python3/lib/libFreeCADApp.so+0x2c
#9  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128008) [0x7fd2c06ef008]
#10  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x7fd2c06a6adc]
#11  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dbc) [0x7fd2c0642a1c]
#12  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7fd2c07893af]
#13  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe) [0x7fd2c07843de]
#14  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x20c4cd) [0x7fd2c07d34cd]
#15  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyRun_StringFlags+0x79) [0x7fd2c07d44f9]
#16  0x7fd2c0ca3e5c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib/freecad-python3/lib/libFreeCADBase.so+0x6c
#17  0x7fd2c1725f77 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x97
#18  0x7fd2c1726117 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0xe7
#19  0x7fd2c1727be7 in Gui::Command::updateActive() from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x47
#20  0x7fd16a486ad5 in finishFeature(Gui::Command const*, App::DocumentObject*, App::DocumentObject*, bool, bool) from /usr/lib/freecad-python3/lib/PartDesignGui.so+0x3c5
#21  /usr/lib/freecad-python3/lib/PartDesignGui.so(+0xe943b) [0x7fd16a49643b]
#22  /usr/lib/freecad-python3/lib/PartDesignGui.so(+0xdfc6f) [0x7fd16a48cc6f]
#23  0x7fd16a48e79a in prepareProfileBased(PartDesign::Body*, Gui::Command*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<void (Part::Feature*, App::DocumentObject*)>) from /usr/lib/freecad-python3/lib/PartDesignGui.so+0x30a
#24  0x7fd16a48f6dc in CmdPartDesignHole::activated(int) from /usr/lib/freecad-python3/lib/PartDesignGui.so+0xdc
#25  0x7fd2c1725a2f in Gui::Command::_invoke(int, bool) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x26f
#26  0x7fd2c1725e80 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x110
#27  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f1793) [0x7fd2bf597793]
#28  0x7fd2c0048be6 in QAction::triggered(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#29  0x7fd2c004b8fc in QAction::activate(QAction::ActionEvent) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xac
#30  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x26408a) [0x7fd2c014708a]
#31  0x7fd2c01471e7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
#32  0x7fd2c0243d9e in QToolButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe
#33  0x7fd2c00924ee in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x20e
#34  0x7fd2c004f713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#35  0x7fd2c0057364 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#36  0x7fd2c16f83a8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x98
#37  0x7fd2bf55fe3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#38  0x7fd2c0055e47 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#39  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c8d40) [0x7fd2c00abd40]
#40  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cbfd5) [0x7fd2c00aefd5]
#41  0x7fd2c004f713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#42  0x7fd2c16f83a8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x98
#43  0x7fd2bf55fe3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#44  0x7fd2bf946307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x697
#45  0x7fd2bf91ba2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#46  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x73d6e) [0x7fd2b9109d6e]
#47  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7fd2bd162d3b]
#48  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab2b8) [0x7fd2bd1b82b8]
#49  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7fd2bd1603e3]
#50  0x7fd2bf5b90b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#51  0x7fd2bf55e75b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#52  0x7fd2bf566cf4 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#53  0x7fd2c1684ded in Gui::Application::runApplication() from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x250d
#54  freecad(+0x6d53) [0x5566a0b9cd53]
#55  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7fd2bee5ad90]
#56  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7fd2bee5ae40]
#57  freecad(+0x7ea5) [0x5566a0b9dea5]
jan@jan-XXXXXXXXX:~$ freecad

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-11:
Please try with the latest weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-11-11:
Please re-open if you can reproduce on RC2 or weekly.

closed 2024-11-11 by luzpaz

duinzicht8, 2024-11-12:
I used the RC2 and it didn't crash, but the tool didn't work like in the video,
I also see that there is an error with the "wire join function" in my drawing.

FreeCAD 1.1.0, Libs: 1.1.0devR39153 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.Vertex1
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
<Sketch> SketchObject.cpp(11403): invalid subname ;g1v1;SKT.
Hole002: Wire is not closed.