FreeCAD/FreeCAD #17795: Exportation STL incomplete de POLARPATERNE d'un ADDITIVEHELIX avec une hauteur de 100mm

alainbron, 2024-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Bonjour,
Si l'additivehelix est plus haute que 60mm, l'exportation lst est fracturée.
[image]
[image]

[image]

[image]

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) Snap 908
Build type: Release
Branch: tag: 0.21.2
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

[image]
[image]

[image]

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-12:
@alainbron please translate to English. Can you try this with the latest weekly build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
If there still is an issue, please provide an example file and steps how to reproduce.

alainbron, 2024-11-12:
Thank you
My poor English :+1: 
The first picture show the result product by the no good data in the stl file. The middle gap is the default. The cura slicer compute the same  default.
The second picture show the structure of the shell corps.
The picture 3 show the values used for ADDITIVEHELIX - I have test with other one. And with 2000mm "pas" all is good !!
The forth picture is that FREECAD show ! The strange pixels are at the 'future' stl gap location.
I use the STL mesh format.
Following the example file
[bb_grille.zip](https://github.com/user-attachments/files/17721031/bb_grille.zip)

alainbron, 2024-11-12:
I have test the same part with the last weekly-builds 39175, all is in order.
Thank you
Alain

closed 2024-11-13 by maxwxyz