FreeCAD/FreeCAD #17796: When opening a sketch, parent bodies' visibility is re-enabled, obscuring your view of the sketch

Stokestack, 2024-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When invoking the sketch editor for a particular sketch, the visibility of operations (like padding) on OTHER sketches (not the one you're editing) is re-activated despite your having turned them off. This obscures your view of the sketch you're trying to work on. You have to go back to the model, go to the other sketch's modifier (or operation or whatever you want to call it), and re-disable its visibility.

In the attached video, please note that **the sketch being edited is NOT the one whose pad is being re-enabled**.

https://github.com/user-attachments/assets/2645bcd5-97a3-48f8-af9c-d229edcb52f8

### Full version info

```shell
OS: macOS 15.1
Architecture: arm64
Version: 1.1.0dev.39153 (Git)
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: Dark-modern.qss/Dark behave/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-11-11:
Please check the following preferences:
* Show objects used for external geometry
* Show objects that the sketch is attached to

https://wiki.freecad.org/Sketcher_Preferences#Display

Stokestack, 2024-11-11:
Thanks for the reply.

The culprit turns out to be "Show objects that the sketch is attached to."

If I activate "Show objects used for external geometry," the sketch editor only shows the other _sketch_; not the filled-in object. This is exactly the behavior I want (and, I would think, is generally the most useful in the sketch editor).

So I'm going close this. Great tip, thanks.

closed 2024-11-11 by Stokestack