FreeCAD/FreeCAD #17797: Assembly: Solve failed: Gear radius is zero.

nmschulte, 2024-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a Gears Joint in the Assembly workbench, the Create Joint task initializes with `Radius 1` (`distanceSpinbox`) and `Radius 2` (`distanceSpinbox2`) values of `1.0`.

However, if unmodified, the value of the Joint's `Distance` and `Distance2` properties listed in the Data view sets as `0.0`.  Because of this, errors like `<Assembly> AssemblyObject.cpp(176): Solve failed: Gear radius is zero.` occur.

The problem is obscured to the user by the deceptive form value manipulation (setting/showing `1.0` in the form when the value is `0.0`), and apparently not correspondingly initializing the "real" data property value.

Modifying the values in the task form corrects this behavior; the task form value and the "real" data property value align.

### Full version info

```shell
e661774a29029660819a5981e2b710638e59d86c
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-12:
@nmschulte can you please provide an example file.

nmschulte, 2024-11-12:
> please provide an example file.

@maxwxyz: see the `Gears` joint in `Assembly`.  Also try creating a new one -- Belt joint type should give similar funk; their implementation is shared (they are basically the same ...).

remove the `.zip` suffix: [fc-17797.FCStd.zip](https://github.com/user-attachments/files/17720286/fc-17797.FCStd.zip)

closed 2024-12-09 by chennes