FreeCAD/FreeCAD #17801: Not able to report on FreeCAD Forum

Generalaccountuse, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have tried contacting the email on the webpage, but have got no response.

### Full version info

```shell
I am simply trying to create a log in, and after entering in all the correct information, it denies my account creation and then locks me out after x amount tries.
```

### Subproject(s) affected?

None

### Anything else?

I can submit snapshots if requested.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-12:
@Roy-043 could you take a look at this regarding forum?
Closing as this is no code issue.

closed 2024-11-12 by maxwxyz

Roy-043, 2024-11-12:
@Generalaccountuse Your email address and/or your IP may be on our block list. Try using a different email address and/or a different IP.

Generalaccountuse, 2024-11-13:
@Roy-043 But why? this is the first time I have ever tried to make a log in to try and ask questions on the forum. Why would either of them be blocked? I did try using different email addresses. First with my proton mail, and then with an old gmail account. Different IP? I just dont understand, why would my IP be blocked? I did try with a public WiFi first and then with a personal one, and it still kept giving the same message that I didn't answer "the questions below properly" even though I entered in all correct info. And like I said before, I already tried sending an email to the contact email on the website but have got  no reply yet. I didn't want to come here to waste peoples time, but I have no other alternatives. Being stuck is a real turn off to be using FreeCAD, which is a shame because I really support the concept.

Do you happen to know of how I can reach someone via email, or otherwise, at the Forum?