FreeCAD/FreeCAD #17802: Basic part design pocket operations fail in 1.0 rc2

Outshined, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Creating a simple part design, pocket operation fails.  In the attached file, try to make a pocket with Sketch003 to create a flat, open side in the shape.  My expectation is that it would look like you shaved half of a balloon away, but it doesn't actually do anything in the preview, and hitting OK gets a popup error with text "Removing splitter failed".

Very specifically:
* open the file
* select the part design workbench
* make Body the active body
* select Sketch003
* click Pocket
* select 'Symmetric to Plane' and Type 'Through all'
  * note that the display does not update
* select OK
  *  Note that you get an error dialog "Removing splitter failed"

[part-design-bug.zip](https://github.com/user-attachments/files/17710361/part-design-bug.zip)

### Full version info

```shell
FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311.AppImage
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Outshined, 2024-11-12:
Running on updated Ubuntu 22.04.  :facepalm:

maxwxyz, 2024-11-12:
Workaround: https://github.com/FreeCAD/FreeCAD/issues/15353#issuecomment-2223407106

Same as: https://github.com/FreeCAD/FreeCAD/issues/15353 but for Pocket or https://github.com/FreeCAD/FreeCAD/issues/14943

AIRCAP, 2024-11-12:
cannot reproduce in latest main. can you verify the bug is gone in latest weekly? 
[image]

Outshined, 2024-11-12:
> Workaround: [#15353 (comment)](https://github.com/FreeCAD/FreeCAD/issues/15353#issuecomment-2223407106)
> 
> Same as: #15353 but for Pocket or #14943

Turning off refining worked for this as well, thanks!

Outshined, 2024-11-12:
> cannot reproduce in latest main. can you verify the bug is gone in latest weekly? [...]

Workaround above yielded the same results on the RC2 build, are your settings per the workaround?  (I'll try to hunt down the build you mentioned and test there.)

Outshined, 2024-11-14:
Thank you, downloaded the weekly build and bug still happens with the same drawing.  Note that if I disable refinement (per workaround above) it behaves properly in both the RC and the weekly.

maxwxyz, 2024-11-14:
@Outshined please post full about info

luzpaz, 2025-01-19:
ping @Outshined 👆

Outshined, 2025-01-19:
Derp - apologies for missing the request above.  Pulled the dev version this morning and tried the same drawing/steps.  Error dialog changed to just say "Error" and nothing else.

> OS: Ubuntu Core 22 (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39904 (Git) Snap 1271
Build type: Release
Branch: main
Hash: e22b226866fcf1e4aa733ddfb902041bf2a9910b
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

GeeBee2, 2026-10-26:
I can't reproduce with or without refine.

```
OS: macOS 15.6.1
Architecture: arm64
Version: 26.3.0RC1.20261008 (Git shallow)
Build date: 2026/10/08 14:07:50
Build type: Release
Branch: (HEAD detached at faa2c25b1)
Hash: faa2c25b187f46623145d09c86142d2bb9c89019
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
QPA/File dialog/Color dialog: cocoa/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
OpenGL version/vendor/renderer: 2.1 Metal - 89.4/Apple/Apple M1
```