FreeCAD/FreeCAD #17803: Property of "Points" is displayed and modified in millimeters while using general setting of imperial units

daveinman, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the "Part" worksheet, I am trying to enter values for the "Points" property of a polygon by hand.  I bring up the dialog and have it show me a table of points.  Though I have specified that I will be using "Building US..." for the units used in the application, the points are listed in millimeters.  This means having to manually convert many entries from millimeters to inches by hand - a slow and error prone process.

### Full version info

```shell
[code]
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.68
[/code]
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-11-12:
Are you sure this is about a Part WB object? Maybe you mean the Draft_Wire object instead?

daveinman, 2024-11-12:
> Are you sure this is about a Part WB object? Maybe you mean the Draft_Wire object instead?

I'm new to FreeCAD and I'm sure you're right.  However, I can see the property under "Part" WB.  I appreciate the correction.

Roy-043, 2024-11-12:
I am confused now. Can you attach a file? Just add the .zip extension to FCStd filename.

daveinman, 2024-11-12:
Here is a zip of .FCstd file I'm using:
[test-freecad.zip](https://github.com/user-attachments/files/17723257/test-freecad.zip)

Roy-043, 2024-11-13:
You have a Std_Part in the Tree view, which is not defined in the Part WB. The polygon you mention is indeed a Draft_Wire.

Roy-043, 2025-02-07:
Related: #11459.