FreeCAD/FreeCAD #17804: Cannot copy and paste points in Points property

daveinman, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the "Part" worksheet and having a floor polygon selected, I view its data property "Points".  I click on the dot button to view the data in a dialog and, once it's up and viewed I select "table" to see all of the points.  I can highlight these points and try and copy them, but they will not actually be copied properly such that I can paste them in, say, a spreadsheet table.

### Full version info

```shell
[code]
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.68
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-11-12:
Are you sure this is about a Part WB object? Maybe you mean the Draft_Wire object instead?

maxwxyz, 2024-11-12:
And please try it in the latest weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Can you provide a demo file?

daveinman, 2024-11-12:
I downloaded FreeCAD_weekly-builds-39175-conda-macOS-arm64-py311.dmg to my Mac Book Pro (Apple M2 Pro).  When opening it I get and apple dialog saying: 
“FreeCAD” is damaged and can’t be opened. You should move it to the Trash.

[test-freecad.zip](https://github.com/user-attachments/files/17721640/test-freecad.zip)

maxwxyz, 2024-11-12:
@daveinman here is the workaround: https://github.com/FreeCAD/FreeCAD-Bundle/issues/329#issuecomment-2453605583

daveinman, 2024-11-12:
Thanks.
Okay, I'm running:
OS: macOS 14.5
Architecture: arm64
Version: 1.1.0dev.39175 (Git) Conda
Build type: Release
Branch: main
Hash: 46831f4c86a1679d48f63273b92cd755522e42f9
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * A2plus 0.4.68

It exhibits this bug.  Copying the table of points in FreeCAD, then pasting in an app yields one entry '0'

Roy-043, 2025-01-14:
Completed with #17851.

closed 2025-01-14 by Roy-043