FreeCAD/FreeCAD #17805: While I built the project, the FreeCADGui get an error : LNK1189  library limit of number objects exceeded

liquor2349, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While I built the project, the FreeCADGui get an error : LNK1189  library limit of number objects exceeded. Also the boost get some problem in the "posioning.hpp" line 96, pos cannot call seekops. So I changed it into "return std::fpos_t(pos);". I don't konw whether it is correct. 

### Full version info

```shell
Moreover, I have already add /bigobj in VS 2022 trying to deal with LNK1189, but it doesn't work.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

liquor2349, 2024-11-12:
Maybe I should have explained it a little bit more clearly, but the problem I had at the beginning was
Error C2039 "seekpos": FreeCADBase is not a member of "std::fpos<_Mbstatet>" D:\freecad\LibPack-0.21-V2.10\include\boost\iostreams\positioning.hpp 96
I changed it to return fpos_t(pos); To solve this problem, which led to the present problem

liquor2349, 2024-11-12:
Moreover, if you compiled in windows, VS 2022. It is need to add  "set(CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS ON)" to CmakeList.txt in ./src/Gui, which will lead to lib files not found. Am I right?

luzpaz, 2024-11-23:
> LibPack-0.21  

Is this the libpack to build 0.21.x with? Is there a libpack to build 1.0 with ?

luzpaz, 2025-08-27:
cc @liquor2349 this still an issue?

closed 2025-12-07 by luzpaz