FreeCAD/FreeCAD #17807: STEP import missing Edges

easyw, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

this is a long time issue still partially advanced toward the solution...
In importing Prefs, if I don't enable **step compound merge**, edges shapes will be dropped.
I double checked importing the exported STEP file (which is fine) with CAD Assistant 1.6.0
Test files and discussion at the FC forum
https://forum.freecad.org/viewtopic.php?t=77994
The issue is present with the importing settings added at the [forum post](https://forum.freecad.org/viewtopic.php?p=791488#p791488)

### Full version info

```shell
OS: Windows 10 build 18363
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-12:
Can you upload the files here as .zip? Please also test on the latest weekly version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

easyw, 2024-11-12:
@maxwxyz 
already tested w/ failure w/ latest weekly build.
here the files, from the FC post:
[Composite-obj-CompositeCircle.zip](https://github.com/user-attachments/files/17719291/Composite-obj-CompositeCircle.zip)
[Composite-obj.zip](https://github.com/user-attachments/files/17719292/Composite-obj.zip)
[Composite-obj-Composite.zip](https://github.com/user-attachments/files/17719293/Composite-obj-Composite.zip)

luzpaz, 2024-11-22:
Confirmed by @adrianinsaval on the forum thread

FEA-eng, 2026-05-04:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260429 (Git shallow)
Build date: 2026/04/29 07:07:16
Build type: Release
Branch: grafted,grafted
Hash: a785d40ac4c37f559b0b82fbaf9bf7dc68a468e4
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```